SpinalHDL/VexRiscv #121: Murax on simulator doesn't store values into RAM

yokoyama-flogics, 2020-04-03:
Hello,

I'm enjoying VexRiscv and also SpinalHDL very much.  Thank you for your great software.

However, I have a strange problem when running Murax by simulation.

Followed the explanation in the URL:

https://github.com/SpinalHDL/VexRiscv/README.md (git commit 31d2aaa)

and executed

```bash
$ cd somewhere/VexRiscv
$ sbt "runMain vexriscv.demo.Murax"
$ cd src/test/cpp/murax
$ make clean run
```

In a different terminal, ran your version of openocd_riscv (git commit 2743d9b) as

```bash
$ openocd -f interface/jtag_tcp.cfg -c "set MURAX_CPU0_YAML /home/yokoyama/VexRiscv/cpu0.yaml" -f target/murax.cfg
```

And finally ran gdb (SiFive GNU Embedded Toolchain — v2019.08.0 binary) as

```bash
$ cd somewhere/VexRiscvSocSoftware/projects/murax/demo
$ make
$ /opt/riscv/bin/riscv64-unknown-elf-gdb build/demo.elf 
GNU gdb (GDB) 8.0.50.20170724-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/demo.elf...done.
(gdb) target remote :3333
Remote debugging using :3333
crtStart () at src/crt.S:6
6	  j crtInit
(gdb) monitor reset halt
JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
(gdb) load
Loading section .vector, size 0x130 lma 0x80000000
Loading section .memory, size 0x354 lma 0x80000130
Loading section .data, size 0x4 lma 0x80000484
Start address 0x80000000, load size 1160
Transfer rate: 12 KB/sec, 386 bytes/write.
(gdb) tb main
Temporary breakpoint 1 at 0x800002d4: file src/main.c, line 7.
(gdb) c
Continuing.

Program stopped.
main () at src/main.c:7
7		volatile uint32_t a = 1, b = 2, c = 3;
(gdb) n

Program stopped.
main () at src/main.c:12
12		timer_init(TIMER_A);
```

Strange to say, local variables on stack aren't stored correctly.

```bash
(gdb) p a
$1 = 2785635792
(gdb) p b
$2 = 1
(gdb) p c
$3 = 4035717316
(gdb) n
19		TIMER_INTERRUPT->PENDINGS = 0xF;
```

I also checked by memory dump.

```bash
(gdb) info reg sp
sp             0x80000660	-2147482016
(gdb) p/ &a
$4 = (volatile uint32_t *) 0x80000674
(gdb) p/ &b
$5 = (volatile uint32_t *) 0x80000670
(gdb) p/ &c
$6 = (volatile uint32_t *) 0x8000066c
(gdb) x/32x 0x80000660
0x80000660:	0xf6204973	0xa3f03c89	0x717e8c99	0xf08c28c4
0x80000670:	0x00000001	0xa6096dd0	0xfe7af673	0x6285a8cb
0x80000680:	0xf6fe060e	0x247ed833	0x0000d4d4	0x8000012c
0x80000690:	0x2066adf4	0x6091c3f4	0xc2bdb7b1	0xdfb08152
0x800006a0:	0xc72d93f8	0xf0d9c1a9	0xdfe8f1fb	0x241c140c
0x800006b0:	0x5abe2388	0xfb7afa79	0x0902faf2	0x9291908f
0x800006c0:	0x4589cd10	0x7970665d	0x09d0985f	0xa3896f55
0x800006d0:	0x3a3d4043	0x77c4115f	0x087ef46b	0x03b05d0a
(gdb) 
```

Variables a, b, c and also result are strange even by memory dump.

However, I confirmed that I could directly assign values to variables by gdb.

```bash
(gdb) set {int}0x8000066c=0x123
(gdb) x/4x 0x80000660
0x80000660:	0xf6204973	0xa3f03c89	0x717e8c99	0x00000123
(gdb) 
```

On the other hand, when run the demo code, UART transmitter is working
and the simulator shows the letter 'A' on screen by the following line:

```c++
    UART->DATA = 'A';
```

In other words, writing to peripheral registers look working fine...

By the way, this problem doesn't occur on actual FPGA (Murax on iCE40).  Only occurs on simulator.

My environment is Ubuntu 16.04.6 LTS.

Any suggestion would be highly appreciated.

Regards,

Atsushi Yokoyama

yokoyama-flogics, 2020-04-03:
Just in case, also checked assembly output (beginning part) of the function main().

```asm
void main() {
800002c4:       fd010113                addi    sp,sp,-48
800002c8:       02112623                sw      ra,44(sp)
800002cc:       02812423                sw      s0,40(sp)
800002d0:       03010413                addi    s0,sp,48
        volatile uint32_t a = 1, b = 2, c = 3;
800002d4:       00100793                li      a5,1
800002d8:       fef42223                sw      a5,-28(s0)
800002dc:       00200793                li      a5,2
800002e0:       fef42023                sw      a5,-32(s0)
800002e4:       00300793                li      a5,3
800002e8:       fcf42e23                sw      a5,-36(s0)
        uint32_t result = 0;
800002ec:       fe042623                sw      zero,-20(s0)

        interruptCtrl_init(TIMER_INTERRUPT);
800002f0:       f00207b7                lui     a5,0xf0020
800002f4:       01078513                addi    a0,a5,16 # f0020010 <__global_pointer$+0x7001f388>
```

Regards,

Atsushi

yokoyama-flogics, 2020-04-03:
FYI, Verilator version is 3.918.

Dolu1990, 2020-04-06:
Hi, sorry for the delay.
So normaly i'm using eclipse to debug stuff, and there as far i know, all is fine, but i to check following your procedure to see what is going wrong.

Should have time to do that tomorrow :)

yokoyama-flogics, 2020-04-06:
Thank you for your attention!

I will dig into the problem by myself too.
It will lead to my better understanding of SpinalHDL and also memory architecture around VexRiscv.

Regards,
Atsushi

yokoyama-flogics, 2020-04-07:
I have checked CPU and bus behaviors, **without OpenOCD connection**, on gtkview and confirmed that memory write accesses are fine.

Following is memory write to 0x8000066c.
```asm
800002e8:       fcf42e23                sw      a5,-36(s0)
```

[image]

And the following is read access to the same address.
```asm
800003a4:       fdc42783                lw      a5,-36(s0)
```

[image]

So my current conclusion is, the problem is not due to CPU nor bus behaviors.

I think I need to dive into interactions between OpenOCD and JTAG.

Regards,
Atsushi

Dolu1990, 2020-04-07:
So basicaly, i updated how the step by step worked in the CPU, but that's broken, i'm fixing it now.

Dolu1990, 2020-04-07:
Fixed in dev and master :)

Also for the sim of murax, there is also sbt "test:runMain vexriscv.MuraxSim"

Thanks for the issue !

closed 2020-04-07 by Dolu1990

yokoyama-flogics, 2020-04-08:
Thank you for prompt fix!
Confirmed that problem was resolved.

Regards,
Atsushi