SpinalHDL/VexRiscv #122: Murax/iCE40-hx8k_breakout_board_xip can't build

yokoyama-flogics, 2020-04-22:
Hello,

I'm afraid of frequent questions...

When I followed the explanation below URL, but I saw a build error.  (I'm evaluating VexRiscv commit ddc59bc.)

- https://github.com/SpinalHDL/VexRiscv/tree/master/scripts/Murax/iCE40-hx8k_breakout_board_xip

The error is as following.

```bash
$ cd somewhere/VexRiscv/scripts/Murax/iCE40-hx8k_breakout_board_xip
$ make compile
(cd ../../..; sbt "runMain vexriscv.demo.Murax_iCE40_hx8k_breakout_board_xip")
[info] Loading settings for project vexriscv-build from plugins.sbt ...
[info] Loading project definition from /Users/yokoyama/tmp/VexRiscv/project
[info] Loading settings for project root from build.sbt ...
[info] Set current project to VexRiscv (in build file:/Users/yokoyama/tmp/VexRiscv/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running (fork) vexriscv.demo.Murax_iCE40_hx8k_breakout_board_xip 
[info] [Runtime] SpinalHDL v1.4.0    git head : ecb5a80b713566f417ea3ea061f9969e73770a7f
[info] [Runtime] JVM max memory : 5461.5MiB
[info] [Runtime] Current date : 2020.04.22 10:49:04
[info] [Progress] at 0.000 : Elaborate components
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 0.916 : Elaborate components
[error] Exception in thread "main" scala.NotImplementedError: an implementation is missing
[error]         at scala.Predef$.$qmark$qmark$qmark(Predef.scala:230)
[error]         at spinal.lib.com.spi.ddr.SpiXdrMasterCtrl$XipBus.fromPipelinedMemoryBus(SpiXdrMasterCtrl.scala:309)
[error]         at vexriscv.demo.Murax$$anon$4$$anonfun$8$$anon$6.<init>(Murax.scala:301)
(snip)
```

I'm not confident but the following line of SpinalHDL/lib/src/main/scala/spinal/lib/com/spi/xdr/SpiXdrMasterCtrl.scala (v1.4.0) looks causing the error.

```scala
    def fromPipelinedMemoryBus() = {
      val accessBus = new PipelinedMemoryBus(PipelinedMemoryBusConfig(24,32))
      ???
//      cmd.valid <> (accessBus.cmd.valid && !accessBus.cmd.write)
//      cmd.ready <> accessBus.cmd.ready
//      cmd.payload <> accessBus.cmd.address
//
//      rsp.valid <> accessBus.rsp.valid
//      rsp.payload <> accessBus.rsp.data
      accessBus
    }
```

May I ask if we can evaluate the XIP functionality by the recent VexRiscv releases?

Thanks and Best Regards,
Atsushi Yokoyama

Dolu1990, 2020-05-02:
Hi, 
Sorry for the delay, i was in a VexRiscv SMP rush, finaly got linux to boot on it with a quad core configuration on Arty A7 :)

So, basicaly, i commented that out because i changed the way the Xip interface work. by the past it was word based (you ask a address, it return you 32 bits data), now it is burst based (you specify the address and how many byte you want, and it return those byte one by one). Hmm to restore the functionalty we woudl basicaly just need to add a buffer on the RSP to rebuild a word access from 8 bits beats.

Do you think you can manage it :D ?

yokoyama-flogics, 2020-05-03:
Hello,
Thank you for your reply.

I'm not in a hurry about XiP experimentation.  In addition, burst based XiP looks interesting.  Does it work fine with I-cache?

BTW, running Linux on SMP sounds great.  Please concentrate on the current project!

Regards,
Atsushi

yokoyama-flogics, 2020-05-03:
Hi Dolu1990,

May I have one request!?

When I will evaluate your new XiP implementation (some day), I want to port it to TinyFPGA BX.  [That looks having Adesto Technologies AT25SF081-SSHD on it](https://github.com/tinyfpga/TinyFPGA-BX/blob/master/board/TinyFPGA-BX-Schematic.pdf).  It would be helpful if your implementation is portable and you show which code we should change for different Flash ROMs.

Thanks and Regards,
Atsushi

yokoyama-flogics, 2020-05-23:
Hello Dolu1990,

Now I could implement XiP (execute in place) for TinyFPGA BX and Adesto SPI Flash memory by myself.  Connected SPI ROM to Briey AXI4 crossbar.  (I don't think it's good design, but I will put the code onto my GitHub repo in the near future.)

My current implementation doesn't use Quad read or SB_IO DDR functionality of iCE40, so it's very slow, but I'm satisfied very much.

Thank you for your support always.  SpinalHDL and Vexriscv are very interesting to learn.  Please close this thread.

Regards,
Atsushi

Dolu1990, 2020-05-23:
Ok :)
Thanks

So right, got SMP to work on https://github.com/enjoy-digital/litex_vexriscv_smp
Now starting to port it on SaxonSoc :)

closed 2020-05-23 by Dolu1990

yokoyama-flogics, 2020-05-25:
Hi,

I want to evaluate running Linux with SMP in the near future.

On the other hand, I have published my own XiP implementation for TinyFPGA BX (iCE40-LP8K).
Just for your information. :blush:

- [VexRiscv (mini) Briey port to TinyFPGA BX](https://github.com/yokoyama-flogics/VexRiscv)

Regards,
Atsushi