SpinalHDL/VexRiscv #123: How do you achieve such low lut counts?

timleijt, 2020-05-18:
@Dolu1990 
Hello,

I'm writing my own RISC-V core in Verilog and am looking for ways to make it smaller since it doesn't fit on my FPGA right now. It uses about 5400 luts for the core itself. This is only the rv32I extension set, nothing else.
I came across your project and noticed how you use significantly fewer luts(731!). So I'm wondering how you got it so small? In my project I noticed about 60% of my luts go to the registers, how did you solve that?

I hope you can explain a little about your project to me and apologies for asking a kinda offtopic question.

Thanks!

Dolu1990, 2020-05-18:
Hi,

Hmm, did you implemented some CSR ? or not yet ?

One thing about Cyclone IV, is that as the FPGA do not support asyncronus reads on memories (no distributed ram), it is realy important to read the register file over one full cycle to allow the synthesis to infer a ram block.

In VexRiscv, with the SYNC register file config, it send the read address to the regfile one cycle before the decode stage, and syncronusly read the regfile to get its value in the decode stage.
That might be the main difference ?

If you switch the SYNC config to ASYNC you will see the occupancy exploiding :)

let's e know how it goes !

timleijt, 2020-05-18:
So you are saying I should move my registers to block ram? Wouldn't that be extremely slow? Also, doesn't that mean I can only read 1 value from registers at a time(per cycle I mean)?

Dolu1990, 2020-05-18:
> So you are saying I should move my registers to block ram? 

Yes

> Wouldn't that be extremely slow

Not realy, you can have two rams (2 * (1r + 1w)), connect both write portes together, and using the two read ports independantly to read read two registers by cycles

timleijt, 2020-05-18:
Okay thanks! I'll modify my design to move registers to block ram and see how it goes.

timleijt, 2020-05-19:
Hello I'm back.

Sorry to ask you this, but I can't get quartus to infer my registers to block ram.
I'm making sure I'm only accessing the registers once a cycle by counting cycles. Here is my entire code: https://pastebin.com/XiH1zYiq
So my question is, can you see why? I thought I just didn't understand how to infer ram to block ram, so I wrote a tiny test to see if I can get that to work, and I did get that to work correctly:
```
module membittest(
	input wire clk,
	output reg [31:0]rd1,
	output reg [4:0]rd1Addr
);

	reg [31:0]registers[0:31];
	reg [7:0]count = 0;
	
	//reg [4:0]rd1Addr = 0;
	
	integer i;
	initial begin
		for (i = 0; i < 32; i = i+1)
		begin
			registers[i] = 0;
		end
	end
	
	always @ (posedge clk)
	begin
		count = count + 1;
		case (count)
			0: begin
				rd1 = registers[rd1Addr];
			end
			1: begin
				rd1 = rd1+1;
				registers[rd1Addr] = rd1;
				rd1Addr = rd1Addr + 1;
				count = -1;
			end
		endcase
	end

endmodule

```
This does move the registers to block ram, however, my riscv core code doesn't even though I seem to be doing largely the same thing.
Can you see what I'm doing wrong?

Thanks!

Dolu1990, 2020-05-19:
Hi,

My guess is that the synthesis tool don't like the registers array read/write inside a case statement.
I already had case where having the registers read/write inside a if(x & y) instead of a if(x) was making the inferation not working. XD

So normaly in the quartus software, in the text editor, there is a icon to open a library of templates, in it should be some patterns to write memory that can be reconized by the synthesis tool as proper memory block.

Good luck ^^

timleijt, 2020-05-19:
> Hi,
>
> My guess is that the synthesis tool don't like the registers array read/write inside a case statement.
> I already had case where having the registers read/write inside a if(x & y) instead of a [...]

Thank you so much! I managed to move the registers and ram to block ram and it finally works! I can now successfully blink an LED using risc-v instructions.
I need to do a lot of optimizing though, currently every instruction is 10 cycles because it reads one byte of ram at a time since it's easier, it also fetches one register at a time. There is definitely room for improvement but right now I'm glad it's running.

So thanks a lot for your help!