SpinalHDL/VexRiscv #125: Question: Is it possible to instantiate 2 cache memories?

RazvanN7, 2020-05-25:
Is it possible to have 2 separate cache memories for the same processor. The idea would be to have 2 cache hierarchies that have different replacement policies. The processor would require 2 sets of load/store intructions (eg. load_using_cache1 and load_using_cache2). I understand that the documentation is a WIP for the dbuscached plugin, but I would be interested if it is possible to have such a separation.

Thanks!

Dolu1990, 2020-05-25:
Hi, 
So currently in the DBusCachedPlugin it isn't implemented, but that would be technicaly fesable. Or maybe it could be a single cache with some special CSR to switch mode ?

RazvanN7, 2020-05-25:
Thank you for the quick answer. 

> So currently in the DBusCachedPlugin it isn't implemented, but that would be technicaly fesable
Any idea if there is a timeline for the DBusCachedPlugin or when it would approximately be implemented?

> Or maybe it could be a single cache with some special CSR to switch mode ?

Hm, interesting idea, but I'm not sure it will work since I want the data to be separate. What goes into 1 cache should never get into the other.

Dolu1990, 2020-05-25:
> Any idea if there is a timeline for the DBusCachedPlugin or when it would approximately be implemented?

Hmm, it could kind of simple, as it can reuse the same cache design two times, and just mask them to select the right one.

I would say, it could be about one day of work ?

Would you like to try ?