SpinalHDL/VexRiscv #126: Regression Test problem

jasonlee1001, 2020-05-26:
after executing 
  sbt "testOnly vexriscv.TestIndividualFeatures"
It runs many ISA, Dhrystone, Coremark tests, but finally fails with following message:

../../src/test/cpp/regression/../../resources/VexRiscvRegressionData/sim/zephyr/tests_kernel_context_rv32i.hex not found
makefile:270: recipe for target 'run' failed
make: *** [run] Segmentation fault (core dumped)

Does anyone knows how to solve this ?
"tests_kernel_context_rv32i.hex" not exist in project directories.

Dolu1990, 2020-05-26:
Hi, when you cloned the repo, did you cloned it recursively ? (--recursive)

jasonlee1001, 2020-05-27:
Thanks Dolu1990 !
After clone recursively, the .hex file is present.

closed 2020-05-27 by jasonlee1001