SpinalHDL/VexRiscv #127: Error when running the verilator simulation of the Briey SoC

wenwang3122, 2020-05-31:
Hi!

When I run the simulation of Briey SoC following instructions from: https://github.com/SpinalHDL/VexRiscv#briey-soc

When I do: 
```
cd VexRiscv/src/test/cpp/briey
make clean run
```

I have the following errors:
"
../main.cpp: In constructor ‘BrieyWorkspace::BrieyWorkspace()’:
../main.cpp:404:27: error: cannot convert ‘SData* {aka short unsigned int*}’ to ‘CData* {aka unsigned char*}’ in assignment
   sdramIo->DQ_writeEnable = &top->io_sdram_DQ_writeEnable;
                           ^
../main.cpp:413:39: error: ‘class VBriey’ has no member named ‘Briey’
   axiClk->add(new VexRiscvTracer(top->Briey->axi_core_cpu));
                                       ^
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-parentheses-equality’
VBriey.mk:72: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/wen/VexRiscv/src/test/cpp/briey/obj_dir'
makefile:54: recipe for target 'compile' failed
make: *** [compile] Error 2

What I did:
I installed all the dependencies except for "libts-dev" as it shows "E: Unable to locate package libts-dev".
"

Question:
Are the errors related to not installing this dependency? Or the error is sitting somewhere else?

Thanks!
Wen

[error.log](https://github.com/SpinalHDL/VexRiscv/files/4708650/error.log)

Dolu1990, 2020-06-01:
Hi :)

I think this is fixed in dev, you tried in the master branch ?

Dolu1990, 2020-06-01:
I merged the dev fix into master, you can pull master and test, that should be fine ^^

wenwang3122, 2020-06-01:
Hi!

Thank you so much for the quick reply.

Yes, I was using master branch. 

After pulling your changes, now this error goes away:
"
../main.cpp: In constructor ‘BrieyWorkspace::BrieyWorkspace()’:
../main.cpp:404:27: error: cannot convert ‘SData* {aka short unsigned int*}’ to ‘CData* {aka unsigned char*}’ in assignment
sdramIo->DQ_writeEnable = &top->io_sdram_DQ_writeEnable;
^
"

However, the following error remains:
"
../main.cpp: In constructor ‘BrieyWorkspace::BrieyWorkspace()’:
../main.cpp:413:39: error: ‘class VBriey’ has no member named ‘Briey’
   axiClk->add(new VexRiscvTracer(top->Briey->axi_core_cpu));
                                       ^
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-parentheses-equality’
"

I tried both master and dev branches, they give me the same compilation results.

Thanks,
Wen

Dolu1990, 2020-06-01:
Hmm, that might be because of the verilator version, which one do you have ?
Idealy it should be 4.x, already had issues with 3.9x.

Let's me know how it goes ^^

wenwang3122, 2020-06-08:
I was using Verilator 3.8x, now upgraded to 4.036, the simulation of Briey works now. 

Thank you very much for the help!

closed 2020-06-08 by wenwang3122