SpinalHDL/VexRiscv #129: Question about interrupts (CsrPlugin)

nickolasB1983, 2020-06-02:
Hi!

I'm trying to make external interrupts work with Murax configuration.
The problem is some of the generated interrupts are lost and the CPU doesn't jump to trap routine. 
My knowledge of spinalHDL is quite poor. I'm just beginning to learn it. Most likely I configured the csr plugin in a wrong way, but unfortunately I can't understand what am I doing wrong.

As far as I can tell jumping to the trap routing depends on 3 conditions:  
1. Interrupts should be enabled, 
2. Interrupt condition should occure
3. The pipline should be empty?
It seems like the 3rd and 2nd condition may not simultaniously happen. The interrupt event may last for 1 or 2 cycles. But the 3rd condition may not hold during these cycles.

Apparently the interrupt condition is not saved in the system until the 3rd condition happen and this leads to a missed interrupt. 

Am I getting it wrong?

Dolu1990, 2020-06-02:
Hi,

> 1) Interrupts should be enabled,

Yes, and you need to enable two flags for that in the CPU, mstatus.mie and mie.xxx

> 3) he pipline should be empty?

Yes so basicaly, the interrupt line should stay high, until the CPU clear the interrupt.
Are you doing some single cycle pulse as interrupts ?

nickolasB1983, 2020-06-02:
Thank you for your response,

Ok, I get it. However in this case I'm not sure how to prevent multiple entering of the ISR. What should be the correct sequence in this case ?
Save context -> Enter ISR -> disable interrupt in mie register for example -> perform necessary operations -> enable interrupt -> return from ISR -> restore context.

Is that correct?

Also, what is the minimum length of the interrupt signal to be registered for sure?

nickolasB1983, 2020-06-02:
Thanks all for help.
It is clear now

closed 2020-06-02 by nickolasB1983