SpinalHDL/VexRiscv #130: Debug problem

PedroCarvalho64, 2020-06-08:
Hi hope everyone is alright. :)

I've been trying to use the debug plugin with the JTAG, when I launch my program without breakpoints the program executes without no problem but when I put a breakpoint I can use step by step but for example in a for loop the increment variable is not incrementing. I've tried to read the memory and write into memory it works without no problem but when I use gdb with step by step seems to not work... The example I used to use the debug plugin is from **VexRiscvAvalonWithIntegratedJtag** and I used the MURAX configuration for OpenOCD.
Is there something wrong with my configuration?
```scala
object GenCustomWithJTAG
{
    def main(args: Array[String]) 
    {
        val report = SpinalVerilog
        {
            //CPU configuration
            val cpuConfig = VexRiscvConfig(
                plugins = List
                (
                new IBusSimplePlugin(
                    resetVector = 0x00000000l,
                    cmdForkOnSecondStage = false,
                    cmdForkPersistence = false,
                    prediction = NONE,
                    catchAccessFault = false,
                    compressedGen = false
                ),
                new DBusSimplePlugin(
                    catchAddressMisaligned = false,
                    catchAccessFault = false
                ),
                new CsrPlugin(CsrPluginConfig.all(0x00000000l)),
                new DecoderSimplePlugin(
                    catchIllegalInstruction = false
                ),
                new RegFilePlugin(
                    regFileReadyKind = plugin.SYNC,
                    zeroBoot = false
                ),
                new IntAluPlugin,
                new SrcPlugin(
                    separatedAddSub = false,
                    executeInsertion = false
                ),
                new FullBarrelShifterPlugin(earlyInjection = false),
                new HazardSimplePlugin(
                    bypassExecute           = false,
                    bypassMemory            = false,
                    bypassWriteBack         = false,
                    bypassWriteBackBuffer   = false,
                    pessimisticUseSrc       = false,
                    pessimisticWriteRegFile = false,
                    pessimisticAddressMatch = false
                ),
                new BranchPlugin(
                    earlyBranch = false,
                    catchAddressMisaligned = false
                ),
                new DebugPlugin(ClockDomain.current.clone(reset = Bool().setName("debugReset"))),
                new YamlPlugin("cpu0.yaml")
                )
            )
            val cpu = new VexRiscv(cpuConfig)
            cpu.rework 
            {
                for (plugin <- cpuConfig.plugins) plugin match 
                {
                    case plugin : DebugPlugin   => plugin.debugClockDomain 
                    {
                        plugin.io.bus.setAsDirectionLess()
                        val jtag = slave(Jtag())
                            .setName("jtag")
                        jtag <> plugin.io.bus.fromJtag()
                    }
                    case _  =>
                }
            }
            cpu.setDefinitionName("VexRiscVJTAG")
            cpu
        }
    }
}
```

Dolu1990, 2020-06-09:
Hi,

The configuration seem ok. To be sure, which version or branch of vexriscv are you using ? Because steps was broken a few months ago, fixed now.

PedroCarvalho64, 2020-06-09:
Yes, I've tested now with the current version and everything is working. It's been a while since I've pulled from VexRiscv because everything was working fine ^^.
Thank you.

closed 2020-06-09 by PedroCarvalho64

fatalfeel, 2021-10-10:
> Yes, I've tested now with the current version and everything is working fine. It's been a while since I've pulled from VexRiscv because everything was working fine ^^. Thank you.

do you use external JTAG like jlink , ARM-JTAG-20-10 OLIMEX.
I want to try debug

fatalfeel, 2021-10-10:
> > > Yes, I've tested now with the current version and everything is working fine. It's been a while since I've pulled from VexRiscv because everything was working fine ^^. Thank you.
> > 
> > 
> > do you use external JTAG like jlink , ARM-JTAG-20-10 OLIMEX. I want to try debug
> 
> Yep, I used a JLink JTAG

sound a good news I want to try in BusBlaster Jtag debug Murax.
But I dout why the hardwareBreakpointCount in Murax.scala is 0
How can you set breakpoint hit on crt.S or hello_world.c, plz help

Pedroo64, 2021-10-10:
Ok.
- What is the output of OpenOCD?
- When you launch OpenOCD, are you passing tcl/interface/ftdi/dp_busblaster.cfg or tcl/interface/ftdi/dp_busblaster_kt-link.cfg instead of tcl/interface/ftdi/ft2232h_breakout.cfg
- I suppose you're running this [program](https://github.com/SpinalHDL/VexRiscv/blob/68e704f3092be640aa92c876cf78702a83167f94/src/main/c/murax/hello_world/src/main.c). When you load your executable and run without any breakpoint does [this message](https://github.com/SpinalHDL/VexRiscv/blob/68e704f3092be640aa92c876cf78702a83167f94/src/main/c/murax/hello_world/src/main.c#L26) prints in your ttyUSBX?

I never tried to debug with another JTAG interface but I think it's the same way as the JLink. When I debug with the JLink I launch OpenOCD with this line :
```sh
src/openocd -f tcl/interface/jlink.cfg -c 'set MURAX_CPU0_YAML ../VexRiscv/cpu0.yaml' -f tcl/target/murax.cfg
```

fatalfeel, 2021-10-10:
thanks u sir~~~
https://tomverbeure.github.io/2021/07/18/VexRiscv-OpenOCD-and-Traps.html
https://ithelp.ithome.com.tw/articles/10196583
they connect their jtag board outside and debug with gdb

I am not sure in atry A7 is using tcl/interface/ftdi/ft2232h_breakout.cfg

fatalfeel, 2021-10-10:
another question
how does Murax.c know the tap idcode in arty a7?

/root/riscv_cpu/VexRiscv/Murax.v 
_zz_jtag_tap_tdoDr <= 32'h10001fff;

openocd murax.cfg
if { [info exists CPUTAPID] } {
   set _CPUTAPID $CPUTAPID
} else {
  # set useful default
   set _CPUTAPID 0x10001fff 
}

fatalfeel, 2021-10-10:
Murax.v. found tap idcode now its in Jar file systemdebugger.scala

fatalfeel, 2021-10-10:
Ur document link is so good thanks a lot man

Pedroo64, 2021-10-10:
I have an Arty S7 but I use the JTAG externally using by connecting it to the JLink because I use most of the time Lattice FPGAs.

fatalfeel, 2021-10-10:
Many thanks many

fatalfeel, 2021-10-10:
in https://github.com/SpinalHDL/VexRiscv/tree/master/doc/nativeJtag
is my modify correct?

case plugin : DebugPlugin => plugin.debugClockDomain
{
resetCtrl.systemReset setWhen(RegNext(plugin.io.resetOut))
//io.jtag <> plugin.io.bus.fromJtag() //external jtag
//arty internal usb jtag: start
val jtagCtrl = JtagTapInstructionCtrl()
val tap = jtagCtrl.fromXilinxBscane2(userId = 2)
jtagCtrl <> plugin.io.bus.fromJtagInstructionCtrl(ClockDomain(tap.TCK))
//arty internal usb jtag: end
}