SpinalHDL/VexRiscv #131: A problem maybe while do plugin initialization

curliph, 2020-07-06:
I  try to build system based on vexriscv,  build I/D subsystem like below:
```
    val core = new ClockingArea(coreDomain) {
        val vexRiscVConfig = VexRiscvConfig(plugins = vexRiscVPlugins)
        val cpu = new VexRiscv(config = vexRiscVConfig)
        var iBus : Axi4ReadOnly = null
        var dBus : Axi4Shared = null
        for (plugin <- vexRiscVConfig.plugins) plugin match {
            case plugin : IBusSimplePlugin => iBus = plugin.iBus.toAxi4ReadOnly()
            case plugin : DBusSimplePlugin => dBus = plugin.dBus.toAxi4Shared()
            case plugin : CsrPlugin => {
                plugin.externalInterrupt := False
                plugin.timerInterrupt := False
            }
            case _ =>
        }
```
while run build,  got error message :
```
[info] ERROR !
[info] A null pointer access has been detected in the JVM.
[info] This could happen when in your SpinalHDL description, you access an signal which is only defined further.
[info] For instance :
[info]   val result = Bool
[info]   result := a ^ b  //a and b can't be accessed there because they are only defined one line below (Software rule of execution order)
[info]   val a,b = Bool
[info]
[error] Exception in thread "main" java.lang.NullPointerException
[error]         at blinky.Blinky$$anon$3$$anonfun$3.apply(Blinky.scala:108)
[error]         at blinky.Blinky$$anon$3$$anonfun$3.apply(Blinky.scala:107)
[error]         at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at blinky.Blinky$$anon$3.<init>(Blinky.scala:107)
[error]         at blinky.Blinky.<init>(Blinky.scala:102)
[error]         at blinky.BlinkyVerilog$$anonfun$main$1.apply(Blinky.scala:157)
[error]         at blinky.BlinkyVerilog$$anonfun$main$1.apply(Blinky.scala:157)
```
I exactly known what the mean, but just do not known how to fix it.
What's wrong with me?

Thanks,
Franz

franz-git, 2020-07-06:
Seems it's the problem caused by "build.sbt“ file configuration.
I get a copy from "spinalTemplateSbtDependencies” progect,  it works!
But I can not point why all this happen. :)
If anyone know how, please...

Thanks and best regards,

Dolu1990, 2020-07-06:
Hi,

Cool that you sorted it out ^^

How was your previous sbt file ?

franz-git, 2020-07-06:
Here is the original copy:

```
name := "riscv-blinky"

version := "0.1.0"

scalaVersion := "2.11.12"

libraryDependencies ++= Seq(
  "com.github.spinalhdl" % "spinalhdl-core_2.11" % "latest.release",
  "com.github.spinalhdl" % "spinalhdl-lib_2.11" % "latest.release"
)

fork := true

lazy val root = Project("root", file("."))
    .dependsOn(vexRiscV)
lazy val vexRiscV = RootProject(file("../VexRiscv"))
```

Dolu1990, 2020-07-06:
Hooo i think you were missing the dslPlugin in the buildsbt, that's likely the main issue

franz-git, 2020-07-06:
Thanks for your help. 
B.T.W,  SpinalHDL, great work!

closed 2020-07-06 by curliph