SpinalHDL/VexRiscv #132: Slow multiply

shawnl, 2020-07-07:
If the Cortex-M0 can have a slow multiply option (32 cycles I believe), why can't the VexRiscv have it to save space? (I am not sure if there is a multiply only subset of the M option, like the bitmanip stuff has many configurations) Thanks.

tomverbeure, 2020-07-07:
The VexRiscv can have it, is just that nobody has needed it so far. 

Most FPGAs have HW multipliers/DSPs, often with built-in pipeline FFs. So on an FPGA, a DSP-based multiplier will cost the same (or less!) than a slow multiplier.

The RISCV specification doesn't have an option that has a MUL but not a DIV. However, there is an option in GCC that forces it to not generate DIVs, even with the M option is specified.

Dolu1990, 2020-07-07:
Agree, as most VexRIscv targets are FPGA with DSP blocks, it isn't very much used. but it is already implemented.
Basicaly, instead of using the new MulSimplePlugin + new DivPlugin in the configuration, you have to use : 

```scala
          new MulDivIterativePlugin(
            genMul = true,
            genDiv = true,
            mulUnrollFactor = 1,
            divUnrollFactor = 1
          ),
``` 

It will implement mul/div using no DSP blocks, but bit by bit (about 34 cycles for one mul).
Also, you can get faster mul by setting mulUnrollFactor = 4, which will do 4 bits per clock instead of one.

tomverbeure, 2020-07-07:
Oops! I should have known better! :-)

Dolu1990, 2020-07-07:
No worries XD