SpinalHDL/VexRiscv #137: Question: Cacheable address region determined?

vasimr, 2020-09-20:
Hi,
How are the cachable address regions determined? I can't seem to find any setting or documentation / comments describing this. Does every address go to the cache, or is it everything except what's mapped to the StaticMemoryTranslatorPlugin? If so, does that only apply to data?

I would like to setup a configuration where I have two copies of DRAM / On Chip Memory, where one is cachable and the other is not, without using an MMU (if they have different physical addresses, I can manually remap them in an external wrapper).

Thanks in advance!

Dolu1990, 2020-09-20:
Hi, basicaly, the instruction cache always use his cache to access instructions, while the data cache support both cached and uncached access.
So, to know if the access should be done in a cached or uncached manner, the data cache plugin will look for another plugin which implement the https://github.com/SpinalHDL/VexRiscv/blob/95237b23ea2d658cb3e0aa77680ca2851ef5d882/src/main/scala/vexriscv/Services.scala#L91 
elaboration interface to provide him a memory translation port.
If you don't use the mmu, you have to use https://github.com/SpinalHDL/VexRiscv/blob/6fc5406901fcd34d2fa166761a588fea1b38cddf/src/main/scala/vexriscv/plugin/StaticMemoryTranslatorPlugin.scala to provide this functionality (as you did)
And that plugin is configured via "ioRange : UInt => Bool" to be able to decode an address into IO region yes/no.

So, if you give as parameter   address => address < 0x1000    That mean that all address smaller than 0x1000 will be an io region.

You can combine expression to have more complicated io region definitions.

 address < 0x1000 ||  address === 0x9000 for example
or address(31 downto 28) === 0xF   would mean that all the address where the 4 MSB are set is io region.

vasimr, 2020-09-20:
Thanks for getting back so quickly!
Is there anything special about the "ioRange = true" flag? Will it hurt performance when accessing something like an on chip RAM?

If there is no way to re-direct the instruction address out of the cache, is there a way to manually flush the cache? I'm working on an application that will need to be able to dynamically alter instructions in RAM by an external master, and I don't want to end up in a case where the instruction ended up being cached and is now stale.

Dolu1990, 2020-09-20:
"ioRange = address => True"
Will mean that all the data access will be in io region => all uncached, yes it will kill performances ^^
Basicaly, io access via the DBUsCachedPlugin are quite slower than the DBusSimplePlugin (uncached), because access are emited in the writeback stage, so no pipelining at all.

For the instruction, that isn't implemented, it only support cached accesses.
To flush the instruction  cache you can do a risc-v "ifence" it will flush the instruciton cache

Just one thing
after the ifence, add 5 nop, to be sure the following instruction are fetched from a flushed icache ^^

vasimr, 2020-09-20:
Thanks! That's a handy instruction.

Is there an alternative to using the static mapper if it kills the performance that much?
Does an uncached access using the MMU have a similar performance hit?

Also, I have several quick (hopefully) questions that I have come across (most are clarification questions). Where would be the best place to ask them?

Dolu1990, 2020-09-20:
> Is there an alternative to using the static mapper if it kills the performance that much?

th estatic mapper is realy light, no worries to have here.

> Does an uncached access using the MMU have a similar performance hit?

Yes + the possibility to have a mmu TLB miss which need to be refilled 

> Also, I have several quick (hopefully) questions that I have come across (most are clarification questions). Where would be the best place to ask them?

There a gitter channel for chat:
https://gitter.im/SpinalHDL/VexRiscv

But git issue are good too ^^

vasimr, 2020-09-20:
Thanks! Then I'll just use the static mapper.

I'll ask the questions in the gitter channel, I wouldn't want to submit tons of issues.

closed 2020-09-22 by vasimr