SpinalHDL/VexRiscv #138: High BRAM usage in DCache

vasimr, 2020-09-22:
Hi,

I am having an issue with Vivado implementing a large number of BRAMs in the DCache. It appears that each byte in the cache is being implemented as an RTL RAM and then replaced with a BRAM for each of them during synthesis. I believe this should be implemented using the byte write enables instead. 

Setting the D-Cache to 4-way at 4KB per way (16 KB total), the resource utilization is 18 BRAMs (I am guessing 32 18Ks for the data + another 4 18Ks for the tags).

Dolu1990, 2020-09-22:
Hi,

> It appears that each byte in the cache is being implemented as an RTL RAM and then replaced with a BRAM for each of them during synthesis.

Yes, that's how the netlist describe it. So basicaly, byte enable are kind of tricky and can't be infered accross all plateform using the same pattern, that's why instead, the default is to emit them using separated ram for each byte, which isn't optimal but at least synthetise in a ram block.

The workaround is to enable ram blackboxification in SpinalHDL and provide a carefuly written ram blackbox.

So to do so, just chain SpinalConfig()`.addStandardMemBlackboxing(blackboxByteEnables).generateVerilog(...)`

Then you can provide the following implementation to vivado : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.1/hardware/synthesis/xilinx/common/RamXilinx.v

vasimr, 2020-09-23:
Thanks for the response. I think I figured out where to add / do what you suggested. However, I now get a generation error:

not found: value blackboxByteEnables

Dolu1990, 2020-09-23:
Which version of VexRiscv are you using ? Should be ok in dev (you will need to clone SpinalHDL in the parent directory of Vex

spinal.core.blackboxByteEnables   is the explicit path. If that's not there it might be that you need to update

vasimr, 2020-12-27:
I was able to correctly generate BRAMs for the cache, however, I am having issues now with generating too many BRAMs elsewhere. For example, the cache tags and register files are being placed in BRAMs, whereas they would be better off in RTLRAMs. RTL RAMs can be forced via a directive in verilog / VHDL. Is there a way to specify a black box module implementation of the memory element per plugin component?

Dolu1990, 2021-01-01:
You can specify the RegFilePlugin as Async instead of Sync in the config.

> Is there a way to specify a black box module implementation of the memory element per plugin component?

No.

maybe your best bet would be to use TCL scripts in the synthesis flow to add those attributes ?