SpinalHDL/VexRiscv #143: adding custom Instruction for multiple clock cycle at execute stage

icewindqz, 2020-10-09:
I went through the "Add a custom instruction to the CPU via the plugin system" (SimdAdd), but this instruction is combinational logic, I'm trying to add sequence logic instruction, eg, require multiple clock cycles at execute stage. (see ref code below) The problem I have is whenever IS_SIMD_ADD(execute stage) rise, rd is writing out 1 clk cycle right after execute stage. Can you help on how to control when to write rd (result) or show me an example of adding sequence logic instruction. Thanks!

      when(execute.input(IS_SIMD_ADD)) {
          execute.output(REGFILE_WRITE_DATA) := rd.asBits
      }

mithro, 2020-10-09:
FYI - @tcal-x

tcal-x, 2020-10-09:
Thanks @mithro 

Hi @icewindqz , I've been using a different approach that supports custom instructions with multi-clock latency, and even variable latency.   But the new custom function units are outside of the CPU itself, so you don't have so much access to CPU internals.

On the `dev` branch, there is a plugin called CfuPlugin.scala.   It is used in for example in https://github.com/SpinalHDL/VexRiscv/blob/dev/src/main/scala/vexriscv/demo/GenSmallAndProductiveCfu.scala .  (I  think there's a typo in that file -- "CFU_FUNCTION_ID_W = 2" should be "CFU_FUNCTION_ID_W = 3").

The resulting VexRiscv has an interface for an external CFU:

```
module VexRiscv (
  ...
  output              CfuPlugin_bus_cmd_valid,
  input               CfuPlugin_bus_cmd_ready,
  output     [2:0]    CfuPlugin_bus_cmd_payload_function_id,
  output     [31:0]   CfuPlugin_bus_cmd_payload_inputs_0,
  output     [31:0]   CfuPlugin_bus_cmd_payload_inputs_1,
  input               CfuPlugin_bus_rsp_valid,
  output              CfuPlugin_bus_rsp_ready,
  input               CfuPlugin_bus_rsp_payload_response_ok,
  input      [31:0]   CfuPlugin_bus_rsp_payload_outputs_0,
  ...
```

The CFU can take as many clocks as you want to respond with the result.   But it can also be combinational and reply in the same clock cycle.

Dolu1990, 2020-10-09:
The iterative shifter plugin is a good example of native multicycle instruction : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/ShiftPlugins.scala#L182

arbitration.haltItself is the key ^^

icewindqz, 2020-10-12:
thanks @tcal-x and @Dolu1990 . I finally got it working, just to confirm by having a CFU/CFU plugin is only an optional in this case, right? cause I got it working without CFU. Do you have any suggestion when is good to have CFU or not to have?

Dolu1990, 2020-10-12:
Right, CFU is optional, and a way to define a generic interface to connect a CPU to multiple custom instruction placed outside the CPU (this fork a new pipepline for the CFU from the execute stage and join futher).

icewindqz, 2020-10-13:
Thanks for your help!

closed 2020-10-13 by icewindqz