SpinalHDL/VexRiscv #144: Privileged Architecture v1.11

lindemer, 2020-10-15:
Zephyr RTOS is in the process of adding support for Physical Memory Protection (PMP), which was added to the RISC-V Privileged Architecture in version 1.10. VexRiscv is already supported by Zephyr, but I believe it's based on Privileged Architecture version 1.9. This means Zephyr's new memory protection features will not be compatible. 

Is anyone still actively developing VexRiscv and/or planning to keep it up-to-date with the latest ISA specification?

Dolu1990, 2020-10-15:
First, disclamer : 
To me, the PMP was always a optional feature, and i never keept track of the actual privileged architecture version, but implemented what was usefull / required for software to run (FreeRTOS, Zephyr, OpenSBI, Linux)

Do you know what would be the minimal number of PMP entry required by zephyr ?

> Is anyone still actively developing VexRiscv and/or planning to keep it up-to-date with the latest ISA specification?

I'm doing most of the dev on VexRiscv, basicaly, implementing a PMP shouldn't be complicated, for instance the MMU implemention is only 160 lines of code (https://github.com/SpinalHDL/VexRiscv/blob/6fc5406901fcd34d2fa166761a588fea1b38cddf/src/main/scala/vexriscv/plugin/MemoryTranslatorPlugin.scala)

but there are a few issues : I do not personnaly have any usage of the PMP, which mean i would not be able to actively use it / test it on real cases nor have a particular motivation to implement it. also i prefer to avoid implementing unused features for the healte of the project.

But if someone is interested and has personnal usage, i can help into the task with pointers and info :)
The plugin system of Vex should make the addition of such feature smooth and clean.

> up-to-date with the latest ISA specification?

In FPGA, the pursue of implementing all features of the spec can have a negative value, as it increase the CPU area and potentialy reduce FMax. All features which isn't vital should be implemented as optional, even if the spec say it is mandatory.

lindemer, 2020-10-15:
> Do you know what would be the minimal number of PMP entry required by zephyr ?

At minimum you need 3 entries to enable userspace threads, but if you want to have shared memory between threads you need 1 additional entry per region.

> But if someone is interested and has personnal usage, i can help into the task with pointers and info :)
The plugin system of Vex should make the addition of such feature smooth and clean.

Okay, great. I'll take a closer look at the existing plugins SpinalHDL documentation before I promise anything :)

Dolu1990, 2020-10-15:
> At minimum you need 3 entries to enable userspace threads, but if you want to have shared memory between threads you need 1 additional entry per region.

Ach, and the Zephyr port make it mandatory ?

> Okay, great. I'll take a closer look at the existing plugins and SpinalHDL documentation before I promise anything :)

Let's me know how things turn ^^

lindemer, 2020-10-15:
> Ach, and the Zephyr port make it mandatory ?

Oh no it's just going to be an optional feature for boards that have enough PMP registers. LiteX/VexRiscv aren't effected. (At the moment I think the only supported board is HiFive1.)

closed 2020-10-15 by lindemer

lindemer, 2020-10-20:
@Dolu1990 Just so you know I'm getting started on a PMP plugin now :) It's been a while since I've worked on a processor design so I can't give you a time estimate yet.

There's three different ways to do PMP addressing (NA4, NAPOT and TOR), but only the first two are required for Zephyr userspace. I don't have a use case for TOR addressing and it's much slower to compute than NAPOT (at least on the HiFive1 Rev B) so I may not implement support for it.

Dolu1990, 2020-10-20:
Hoo i didn't even know there was multiple PMP addressing XD

Thanks for the info ^^