SpinalHDL/VexRiscv #152: unable to run simulation with Briey SOC

icewindqz, 2020-11-30:
**trying to following the readme on Briey SOC demonstration, When trying to get these dependencies: I got this error** 
To run the verilator simulation of the Briey SoC, which can then be connected to OpenOCD/GDB, first get these dependencies:
sudo apt-get install build-essential xorg-dev libudev-dev libts-dev libgl1-mesa-dev libglu1-mesa-dev libasound2-dev libpulse-dev libopenal-dev libogg-dev libvorbis-dev libaudiofile-dev libpng12-dev libfreetype6-dev libusb-dev libdbus-1-dev zlib1g-dev libdirectfb-dev libsdl2-dev
E: Unable to locate package libts-dev   (should I ignore it?)

**when go in src/test/cpp/briey and run the simulation with**
../main.cpp:255:22: fatal error: SDL2/SDL.h: No such file or directory
compilation terminated.
VBriey.mk:66: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/ice/proj/riscv_soc1/vexriscv/src/test/cpp/briey/obj_dir'
makefile:54: recipe for target 'compile' failed
make: *** [compile] Error 2

any idea how to fix it.

icewindqz, 2020-11-30:
after skip the libts-dev package, im able to get it work.

Dolu1990, 2020-11-30:
Hi,

Ok cool, i will remove the libts-dev from the readme then ^^
Thanks

closed 2020-11-30 by Dolu1990