SpinalHDL/VexRiscv #153: Add a new pipeline stage to VexRiscv

genesys2, 2020-12-01:
Hi,

I am trying to add a new pipeline stage between decode and execute to Vexriscv. In my understanding this should be possible by declaring a new stage in VexRiscv.scala and then writing a plugin to populate the stage with logic. The first step was to insert a dummy stage without any logic:

```  
  //Define stages
  def newStage(): Stage = { val s = new Stage; stages += s; s }
  val decode    = newStage()
  val dummyStage = newStage()
  val execute   = newStage()
  val memory    = ifGen(config.withMemoryStage)    (newStage())
  val writeBack = ifGen(config.withWriteBackStage) (newStage())

  def stagesFromExecute = stages.dropWhile(_ != execute)
  ....
  dummyStage.arbitration.flushNext.noBackendCombMerge
```

This seems to work mostly fine, in the Traces you can see the dummy stage propagating the signals etc. However, the regression tests fail due to a false RegFile write, especially the signal `lastStageRegFileWrite_payload_data` seems to be false. Do you know why this is and where I need to make further changes? 

Thanks a lot in advance!

Dolu1990, 2020-12-01:
Hi

Which VexRiscv configuration are you using ?
It is quite possible that some plugin do the assumption that execute follow decode.

All test fails ?

genesys2, 2020-12-02:
I was trying GenFull so that probably explains why. Do you have a rough idea how many of such dependecies there are and where they are located? Do you think it is possible to add a stage with reasonable effort?

Very few tests succeed, mostly the problem seems to be a RegFile write. So overall only 35 tests succed.

Dolu1990, 2020-12-02:
Hmm i think one big thing which is actualy a bit too much hardcoded in VexRiscv, is the HazardSimplePlugin, you can see in it it use explicit stages names to check.

One question is why do you want to add one stage between decode and execute ?
(Maybe there could be workarounds in the fetch stages)

genesys2, 2020-12-02:
I don't actually need another pipeline stage that does something meaningful, I just want to measure the performance impact of an ISA extension with a longer pipeline than 5 stages. Therefore, the idea was to add dummy stages to the design and thus emulate the behavior of a longer pipeline. Ideally, the dummy stages would be after decode and before branches are resolved. 

If you're intrested, this is the project I am working on: https://arxiv.org/pdf/2007.15919.pdf

Dolu1990, 2020-12-02:
hmm, there would be a few workaround, basicaly, in the cacheless configuration you can add as many stage in the fetch with a integer parameter in the IBusSimplePlugin.

The IBusCachedPlugin also can be configured to range from 1 to 4 stages.
Hopefully that could help ?
BranchPlugin can also be set to branch in the execute or in the Memory stage.

genesys2, 2020-12-03:
Okay, thank you for your help! Probably the IBusCachedPlugin is a good approach, especially in combination with the early branch option.

Thank you! :)

closed 2020-12-03 by genesys2