SpinalHDL/VexRiscv #154: loading firmware for Briey

icewindqz, 2020-12-02:
i walked through Briey SOC in the readme, I'm able the get it running. I saw VexRiscv core tried to read instruction from 8000_0000 with some data. 
Question 1: I'm wondering how the firmware is loaded in this example, it seems the firmware is pre-load into on chip ram via backdoor. 
Question 2: What if I need to load the firmware through Jtag interface, and boot from on chip ram, do you have an example?

Dolu1990, 2020-12-02:
Hi,
> Question 1: I'm wondering how the firmware is loaded in this example, it seems the firmware is pre-load into on chip ram via backdoor.

For dev, it is loaded via jtag. For standalone boot, the binarry is loaded into the onchipram via the FPGA bitstream itself, most FPGA support that.

> Question 2:

Basicaly, just connect with openocd (see README) : 
src/openocd -f tcl/interface/jtag_tcp.cfg -c "set BRIEY_CPU0_YAML /home/spinalvm/Spinal/VexRiscv/cpu0.yaml" -f tcl/target/briey.cfg

Then you can use GDB to load the binary in it (See readme of riscv_openocd fork in SpinalHDL org)

icewindqz, 2020-12-06:
Thanks, so BrieyWithMemoryInit is an example of loading binary file into onchipram for simulation.
The openocd is also working fine. 
Just a question, in readme section of "Interactive debug of the simulated CPU via GDB OpenOCD and Verilator", it says "Run a GDB session with YourRiscvToolsPath/bin/riscv32-unknown-elf-gdb ", but it seems we only have riscv64-unknown-elf-gdb tool, then I ran it with riscv64-unknown-elf-gdb, and it seems ok, not sure if there is any potential issue. Is "riscv32 or riscv64" referring to riscv instruction width, I believe Briey is running on 32 bit.

Dolu1990, 2020-12-08:
> riscv64-unknown-elf-gdb

Right that's confusing ^^
But basicaly the riscv64-unknown-elf-gdb toolchain is build with both 32 bits and 64 bits support ^^
So all fine.

icewindqz, 2020-12-09:
Thanks for clarifying

closed 2020-12-09 by icewindqz