SpinalHDL/VexRiscv #155: Briey no cache config

icewindqz, 2020-12-06:
I tried to modify Briey I/DCache to nocache, I saw there is IBusSimplePlugin was commented out in the Briey.scala.
I just uncommented the IBusSimplePlugin and commented out IBusCachedPlugin session, it doesn't seem to work.
here is the part of error msg: any ideas on how to do it properly? thanks in advance!

[error] ## Exception when compiling 90 sources to /home/ice/proj/riscv_soc1/vexriscv/target/scala-2.11/classes
[error] assertion failed: 
[error]   UartCtrlMemoryMappedConfig(uartCtrlConfig = UartCtrlGenerics(dataWidthMax = 8, clockDividerWidth = 20, preSamplingSize = 1, samplingSize = 5, postSamplingSize = 2), txFifoDepth = 16, rxFifoDepth = 16)
[error]      while compiling: /home/ice/proj/riscv_soc1/vexriscv/src/main/scala/vexriscv/demo/Briey.scala
[error]         during phase: superaccessors
[error]      library version: version 2.11.12
[error]     compiler version: version 2.11.12
[error]   reconstructed args: -bootclasspath /usr/l

Dolu1990, 2020-12-08:
You only get that as a error message ?
That's realy slime if o.

How did you modified it ?

closed 2021-12-24 by icewindqz

corentin-fleury, 2023-03-07:
Is it possible to have an update on this ? I am trying to do the same thing and so far I don't see why it's failing to build.