SpinalHDL/VexRiscv #156: Upgrade to SpinalHDL 1.4.0 broke the LiteX build

lindemer, 2020-12-10:
The LiteX SoC sources build VexRiscv from an old commit which is specified [here](https://github.com/litex-hub/pythondata-auto/blob/4d7119eceb41bcb2b5e745d024e7c8a32215c196/modules.ini#L69). VexRiscv was upgraded to SpinalHDL 1.4.0 [on March 9](https://github.com/SpinalHDL/VexRiscv/commit/95237b23ea2d658cb3e0aa77680ca2851ef5d882) which appears to have broken the LiteX build script. The error is quite vague so I'm not the person to debug this:
```
[info] Running (fork) vexriscv.GenCoreDefault
[info] [Runtime] SpinalHDL v1.4.0    git head : ecb5a80b713566f417ea3ea061f9969e73770a7f
[info] [Runtime] JVM max memory : 2638.5MiB
[info] [Runtime] Current date : 2020.12.10 16:51:17
[info] [Progress] at 0.000 : Elaborate components
[info] ERROR !
[info] A null pointer access has been detected in the JVM.
[info] This could happen when in your SpinalHDL description, you access an signal which is only defined further.
[info] For instance :
[info]   val result = Bool
[info]   result := a ^ b  //a and b can't be accessed there because they are only defined one line below (Software rule of execution order)
[info]   val a,b = Bool
[info]
[error] Exception in thread "main" java.lang.NullPointerException
[error] 	at vexriscv.GenCoreDefault$$anonfun$main$1$$anonfun$apply$1$$anonfun$apply$mcV$sp$1.apply(GenCoreDefault.scala:241)
[error] 	at vexriscv.GenCoreDefault$$anonfun$main$1$$anonfun$apply$1$$anonfun$apply$mcV$sp$1.apply(GenCoreDefault.scala:235)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 8 s, completed Dec 10, 2020 4:51:17 PM
```

To reproduce this error:
1. Clone the [VexRiscv sources for LiteX](https://github.com/litex-hub/pythondata-cpu-vexriscv).
2. Replace `file("ext/VexRiscv")` [this line](https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/master/pythondata_cpu_vexriscv/verilog/build.sbt) with `uri("https://github.com/SpinalHDL/VexRiscv.git#95237b23ea2d658cb3e0aa77680ca2851ef5d882")`.
3. Run `sbt clean reload && make && sbt "runMain vexriscv.GenCoreDefault"` from that directory.

The [previous commit to master](ttps://github.com/SpinalHDL/VexRiscv.git/commit/0c7556ad7fc874e296805cad856434d00b536005) does not have this problem.

Dolu1990, 2020-12-11:
Fixed with https://github.com/litex-hub/pythondata-cpu-vexriscv/commit/e93dd361ef3ae6341dbd1dee07e047ad67439474#diff-19f6bff0c9bcd2fcbc8f3f174db1dc25c5befeb9caaf5edc1a6ca39788f55de0R12

Basicaly, SpinalHDL 1.4.0 require a scala compiler plugin to be enabled in the build.sbt :)

lindemer, 2020-12-11:
Thanks :) That fixed the build error but I'm getting a new one. I created [a draft PR](https://github.com/litex-hub/pythondata-cpu-vexriscv/pull/9) in the LiteX cores repository detailing the error.

nickoe, 2021-05-03:
@lindemer Shouldn't this issue be closed by now?

closed 2021-05-04 by lindemer