SpinalHDL/VexRiscv #157: Number of Pipeline Stages in Different Configurations

rlipperts, 2020-12-11:
Good Morning,

In my master's thesis I'm developing a comparison framework for SoC and use LiteX SoCs and the Murax as examples. To describe the different VexRiscv configs used, I am currently building a table that outlines their features. However, I am stuck at determining the number of pipeline stages.

The readme describes VexRiscv as it supports 2-5+ stages. [VexRiscv.scala](https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/VexRiscv.scala)  always adds the two minimal stages 
```
val decode    = newStage()
val execute   = newStage()
```
and per default adds two more
```
var withMemoryStage = true
var withWriteBackStage = true
```
From the [LiteX Configs](https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/master/pythondata_cpu_vexriscv/verilog/src/main/scala/vexriscv/GenCoreDefault.scala#L67) I can see that certain plugins include additional stages. With four stages identified, it seems as if the Fetcher plugin introduces further stages, but I do not understand how exactly I can determine the final number.

In summary, I need to know how many stages the different configurations have. I'd like to understand how the different configurations and plugins influence it.

Thank you for your help!

Dolu1990, 2020-12-11:
> it seems as if the Fetcher plugin introduces further stages

Yes that's right. The fetcher plugin can expand the pipeline outside the VexRiscv pipeline API.

Basicaly, both IBusCachedPlugin and iBusSimplePlugin are using https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/Fetcher.scala#L14 which implement those additional pipeline stage and how to inject the instruction into the pipeline.

Number of stages added by the fetcher = cmdToRspStageCount + injectorStage

Then you can look into the IBusCachedPlugin for instance how it is called : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/IBusCachedPlugin.scala#L44

So one thing to keep in mind, is that from an architectural point of view, the first stage added by the fetcher do not realy count, as it is basicaly holding the PC state.
So cmdToRspStageCount + injectorStage - 1

rlipperts, 2020-12-14:
Thank you for your quick response :)

So the VexRiscv config used by Murax has the Decode and Execute state plus MemoryStage and Writeback stage as it does not override the defaults. 

It [configures the IBusSimplePlugin](https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/IBusSimplePlugin.scala#L244) to have cmdToRspStageCount = [1](https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/IBusSimplePlugin.scala#L230) + [1](https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/Murax.scala#L74). But there is also the [Injector](https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/IBusSimplePlugin.scala#L232) as third  stage which results in 4 + (3 - 1) = 6 stages for Murax. 

Is this Correct?

Dolu1990, 2020-12-16:
Yes this seems correct to me, that was likely a way to relax the timings.

rlipperts, 2020-12-16:
Okay, then I have all i need.
Thank you again for your assistance.

closed 2020-12-16 by rlipperts