SpinalHDL/VexRiscv #160: How to generate the Briey SoC Hardware using DDR3 MT41K128M16JT on Arty7-35T?

ngoctuansqtt, 2021-01-12:
Hello, I am new with both SpinalHDL and RISC-V. I have compiled Murax Soc using internal RAM on Arty7 and they work well. Now I want to use external SDRAM DDR3, I try to edit the Briey.scala as follows:

`//Arty7-35T
object Briey_arty{
  def main(args: Array[String]) {
    object MT41K128M16JT {
      def layout = SdramLayout(
        bankWidth   = 3,
        columnWidth = 10,
        rowWidth    = 14,
        dataWidth   = 16
      )	  
    }
    val config = SpinalConfig()
    config.generateVerilog({
      val toplevel = new Briey(BrieyConfig.default.copy(sdramLayout = MT41K128M16JT.layout))     
      toplevel
    })
  }
}`

But it  generates the incompatible DDR3 interface. Could you please show me how to correct it?

This is a part of Briey.v
`module Briey (
      input   io_asyncReset,
      input   io_axiClk,
      input   io_vgaClk,
      input   io_jtag_tms,
      input   io_jtag_tdi,
      output  io_jtag_tdo,
      input   io_jtag_tck,
      output [13:0] io_sdram_ADDR,
      output [2:0] io_sdram_BA,
      input  [15:0] io_sdram_DQ_read,
      output [15:0] io_sdram_DQ_write,
      output  io_sdram_DQ_writeEnable,
      output [1:0] io_sdram_DQM,
      output  io_sdram_CASn,
      output  io_sdram_CKE,
      output  io_sdram_CSn,
      output  io_sdram_RASn,
      output  io_sdram_WEn,
      input  [31:0] io_gpioA_read,
      output [31:0] io_gpioA_write,
      output [31:0] io_gpioA_writeEnable,
      input  [31:0] io_gpioB_read,
      output [31:0] io_gpioB_write,
      output [31:0] io_gpioB_writeEnable,
      output  io_uart_txd,
      input   io_uart_rxd,
      output  io_vga_vSync,
      output  io_vga_hSync,
      output  io_vga_colorEn,
      output [4:0] io_vga_color_r,
      output [5:0] io_vga_color_g,
      output [4:0] io_vga_color_b,
      input   io_timerExternal_clear,
      input   io_timerExternal_tick,
      input   io_coreInterrupt);`

Dolu1990, 2021-01-13:
Hi,

The sdram controller in Briey is for SDR sdram only. If you want DDR3 support take a look into SaxonSoc, it has a artyA7-35t target.

ngoctuansqtt, 2021-01-24:
Hi,
Thank you for your suggestion.