SpinalHDL/VexRiscv #164: VexRiscv doesn't compile with SpinalHDL 1.4.2

tomverbeure, 2021-01-24:
I want to try some of the new SpinalHDL features, but when I make this change in build.sbt of the VexRiscv repo:

```
-val spinalVersion = "1.4.0"
+val spinalVersion = "1.4.2"
```

I get the following failure:

```
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[info] Compiling 76 Scala sources to /home/tom/projects/VexRiscv/target/scala-2.11/classes ...
[error] /home/tom/projects/VexRiscv/src/main/scala/vexriscv/ip/DataCache.scala:71:27: overloaded method value apply with alternatives:
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter,invalidation: spinal.lib.bus.bmb.BmbInvalidationParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (addressWidth: Int,dataWidth: Int,sourceWidth: Int,contextWidth: Int,lengthWidth: Int,alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.Kind,alignmentMin: Int,accessLatencyMin: Int,canRead: Boolean,canWrite: Boolean,canExclusive: Boolean,maximumPendingTransaction: Int)spinal.lib.bus.bmb.BmbParameter
[error]  cannot be applied to (addressWidth: Int, dataWidth: Int, lengthWidth: Int, sourceWidth: Int, contextWidth: Int, canRead: Boolean, canWrite: Boolean, alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.LENGTH.type, maximumPendingTransactionPerId: Int)
[error]   def getBmbParameter() = BmbParameter(
```

Any ideas?

Tom

Dolu1990, 2021-01-24:
Which branch of VexRIscv did you tried ?

tomverbeure, 2021-01-24:
Master.

This is after doing 'sbt clean'.

The latest commit is from last Thursday, so I assumed that this is the main branch and that it is being tested?

tomverbeure, 2021-01-24:
I tried something else:

* VexRiscv dev branch

    This requires a SpinalHDL installation in the ../SpinalHDL directory

* SpinalHDL in that ../SpinalHDL directory, using the 1.4.2 tag
* Compile VexRiscV (`runMain vexriscv.demo.GenSmallest`)

I first get a bunch of warnings:

```
[info] Compiling 258 Scala sources to /home/tom/projects/SpinalHDL/lib/target/scala-2.11/classes ...
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/JtagTap.scala:116:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/JtagTap.scala:122:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/JtagTap.scala:128:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/JtagTap.scala:133:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/JtagTapInstructions.scala:183:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/JtagTapInstructions.scala:188:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/JtagTapInstructions.scala:193:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/lattice/ecp5/JtagTap.scala:140:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/lattice/ecp5/JtagTap.scala:146:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/com/jtag/lattice/ecp5/JtagTap.scala:152:5: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]     area
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/Stream.scala:122:5: match may not be exhaustive.
[warn] It would fail on the following inputs: (false, true, true), (true, false, true), (true, true, true)
[warn]     (m2s, s2m, halfRate) match {
[warn]     ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/bus/regif/RegInst.scala:282:7: Exhaustivity analysis reached max recursion depth, not all missing cases are reported.
[warn] (Please try with scalac -Ypatmat-exhaust-depth 40 or -Ypatmat-exhaust-depth off.)
[warn]       accType match {
[warn]       ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/bus/regif/BusIfAdapter/Axi4BusInterface.scala:1:1: 
[warn] Found names but no class, trait or object is defined in the compilation unit.
[warn] The incremental compiler cannot record the dependency information in such case.
[warn] Some errors like unused import referring to a non-existent class might not be reported.
[warn]     
[warn] package spinal.lib.bus.regif
[warn] ^
[warn] /home/tom/projects/SpinalHDL/lib/src/main/scala/spinal/lib/bus/regif/BusIfAdapter/AxiLite4BusInterface.scala:1:1: 
[warn] Found names but no class, trait or object is defined in the compilation unit.
[warn] The incremental compiler cannot record the dependency information in such case.
[warn] Some errors like unused import referring to a non-existent class might not be reported.
[warn]     
[warn] package spinal.lib.bus.regif
[warn] ^
[warn] there were 21 feature warnings; re-run with -feature for details
[warn] 15 warnings found
[info] Done compiling.
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
```
[... 69 lines cut ...]
[info] Done packaging.
[error] /home/tom/projects/VexRiscv/src/main/scala/vexriscv/ip/DataCache.scala:510:39: value transferBeatCountMinusOne is not a member of spinal.lib.Stream[spinal.lib.bus.bmb.BmbInv]
[error]       val beatCountMinusOne = bus.inv.transferBeatCountMinusOne(p.bytePerLine)
[error]                                       ^
[error] /home/tom/projects/VexRiscv/src/main/scala/vexriscv/ip/fpu/FpuCore.scala:150:21: value swapPayload is not a member of spinal.lib.Stream[FpuCore.this.RfReadInput]
[error]     val output = s1.swapPayload(RfReadOutput())
[error]                     ^
[error] /home/tom/projects/VexRiscv/src/main/scala/vexriscv/ip/fpu/FpuCore.scala:334:36: value takeHigh is not a member of spinal.core.UInt
[error]       val div = input.rs2.mantissa.takeHigh(log2Up(aproxDepth))
[error]                                    ^
[error] /home/tom/projects/VexRiscv/src/main/scala/vexriscv/ip/fpu/FpuCore.scala:335:66: value takeHigh is not a member of spinal.core.UInt
[error]       val sqrt = U(input.rs1.exponent.lsb ## input.rs1.mantissa).takeHigh(log2Up(aproxDepth))
[... cut ...]
[error] /home/tom/projects/VexRiscv/src/main/scala/vexriscv/ip/fpu/FpuCore.scala:498:24: value swapPayload is not a member of spinal.lib.Stream[FpuCore.this.AddInput]
[error]     val output = input.swapPayload(WriteInput())
[error]                        ^
[error] 25 errors found
```

One of the main problems right now, is that the latest official VexRiscv release/tag is v1.0.1, which is 2  years old.
The other issue is that the `master` branch of VexRiscv is still using 1.4.0 in build.sbt, which is more than a year old.

Are you able to compile VexRiscv with 1.4.2? 

Tom

Dolu1990, 2021-01-24:
Hi,

Hmm maybe you have a local version of SpinalHDL 1.4.2 which isn't the
released on ? can you check in your .ivy2/local ?

For me it works :

rawrr@rawrr:~/tmp$ git clone https://github.com/SpinalHDL/VexRiscv.git
--branch master
fatal: destination path 'VexRiscv' already exists and is not an empty
directory.
rawrr@rawrr:~/tmp$ rm -rf VexRiscv
rawrr@rawrr:~/tmp$ git clone https://github.com/SpinalHDL/VexRiscv.git
--branch master
Cloning into 'VexRiscv'...
remote: Enumerating objects: 442, done.
remote: Counting objects: 100% (442/442), done.
remote: Compressing objects: 100% (224/224), done.
remote: Total 12286 (delta 221), reused 372 (delta 185), pack-reused 11844
Receiving objects: 100% (12286/12286), 11.30 MiB | 9.53 MiB/s, done.
Resolving deltas: 100% (7302/7302), done.
rawrr@rawrr:~/tmp$ cd VexRiscv/
rawrr@rawrr:~/tmp/VexRiscv$ sbt compile
[info] Loading global plugins from /home/rawrr/.sbt/1.0/plugins
[info] Loading settings for project vexriscv-build from plugins.sbt ...
[info] Loading project definition from /home/rawrr/tmp/VexRiscv/project
[info] Loading settings for project root from build.sbt ...
[info] Set current project to VexRiscv (in build
file:/home/rawrr/tmp/VexRiscv/)
[info] Executing in batch mode. For better performance use sbt's shell
[info] Compiling 76 Scala sources to
/home/rawrr/tmp/VexRiscv/target/scala-2.11/classes ...
[warn] there were 35 deprecation warnings; re-run with -deprecation for
details
[warn] there were two feature warnings; re-run with -feature for details
[warn] two warnings found
[info] Done compiling.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.protobuf.UnsafeUtil
(file:/home/rawrr/.sbt/boot/scala-2.12.7/org.scala-sbt/sbt/1.2.7/protobuf-java-3.3.1.jar)
to field java.nio.Buffer.address
WARNING: Please consider reporting this to the maintainers of
com.google.protobuf.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal
reflective access operations
WARNING: All illegal access operations will be denied in a future release
[success] Total time: 17 s, completed Jan 24, 2021, 10:57:55 PM
rawrr@rawrr:~/tmp/VexRiscv$

tomverbeure, 2021-01-30:
You tried this:
```
rawrr@rawrr:~/tmp$ rm -rf VexRiscv
rawrr@rawrr:~/tmp$ git clone https://github.com/SpinalHDL/VexRiscv.git
--branch master
Cloning into 'VexRiscv'...
...
```
I did that as well and it works.

But the problem is that the `master` branch of VexRiscv still uses 1.4.0: https://github.com/SpinalHDL/VexRiscv/blob/50a69d8d4a1911b38f622307debfb6a06e794979/build.sbt#L1.

When I simply change that line in `build.sbt` to 1.4.2, I get the failures:

```
tom@thinkcenter:~/projects/VexRiscv.test$ sbt compile
[info] Loading settings for project global-plugins from idea.sbt ...
[info] Loading global plugins from /home/tom/.sbt/1.0/plugins
[info] Loading settings for project vexriscv-test-build from plugins.sbt ...
[info] Loading project definition from /home/tom/projects/VexRiscv.test/project
[info] Loading settings for project root from build.sbt ...
[info] Set current project to VexRiscv (in build file:/home/tom/projects/VexRiscv.test/)
[info] Executing in batch mode. For better performance use sbt's shell
[info] Updating ...
[info] Done updating.
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[info] Compiling 76 Scala sources to /home/tom/projects/VexRiscv.test/target/scala-2.11/classes ...
[error] /home/tom/projects/VexRiscv.test/src/main/scala/vexriscv/ip/DataCache.scala:71:27: overloaded method value apply with alternatives:
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter,invalidation: spinal.lib.bus.bmb.BmbInvalidationParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (addressWidth: Int,dataWidth: Int,sourceWidth: Int,contextWidth: Int,lengthWidth: Int,alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.Kind,alignmentMin: Int,accessLatencyMin: Int,canRead: Boolean,canWrite: Boolean,canExclusive: Boolean,maximumPendingTransaction: Int)spinal.lib.bus.bmb.BmbParameter
[error]  cannot be applied to (addressWidth: Int, dataWidth: Int, lengthWidth: Int, sourceWidth: Int, contextWidth: Int, canRead: Boolean, canWrite: Boolean, alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.LENGTH.type, maximumPendingTransactionPerId: Int)
[error]   def getBmbParameter() = BmbParameter(
[error]                           ^
[error] /home/tom/projects/VexRiscv.test/src/main/scala/vexriscv/ip/InstructionCache.scala:71:27: overloaded method value apply with alternatives:
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter,invalidation: spinal.lib.bus.bmb.BmbInvalidationParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (addressWidth: Int,dataWidth: Int,sourceWidth: Int,contextWidth: Int,lengthWidth: Int,alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.Kind,alignmentMin: Int,accessLatencyMin: Int,canRead: Boolean,canWrite: Boolean,canExclusive: Boolean,maximumPendingTransaction: Int)spinal.lib.bus.bmb.BmbParameter
[error]  cannot be applied to (addressWidth: Int, dataWidth: Int, lengthWidth: Int, sourceWidth: Int, contextWidth: Int, canRead: Boolean, canWrite: Boolean, alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.LENGTH.type, maximumPendingTransactionPerId: Int)
[error]   def getBmbParameter() = BmbParameter(
[error]                           ^
[error] /home/tom/projects/VexRiscv.test/src/main/scala/vexriscv/plugin/DBusSimplePlugin.scala:80:27: overloaded method value apply with alternatives:
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter,invalidation: spinal.lib.bus.bmb.BmbInvalidationParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (addressWidth: Int,dataWidth: Int,sourceWidth: Int,contextWidth: Int,lengthWidth: Int,alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.Kind,alignmentMin: Int,accessLatencyMin: Int,canRead: Boolean,canWrite: Boolean,canExclusive: Boolean,maximumPendingTransaction: Int)spinal.lib.bus.bmb.BmbParameter
[error]  cannot be applied to (addressWidth: Int, dataWidth: Int, lengthWidth: Int, sourceWidth: Int, contextWidth: Int, canRead: Boolean, canWrite: Boolean, alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.LENGTH.type, maximumPendingTransactionPerId: Int)
[error]   def getBmbParameter() = BmbParameter(
[error]                           ^
[error] /home/tom/projects/VexRiscv.test/src/main/scala/vexriscv/plugin/IBusSimplePlugin.scala:77:59: overloaded method value apply with alternatives:
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter,invalidation: spinal.lib.bus.bmb.BmbInvalidationParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (access: spinal.lib.bus.bmb.BmbAccessParameter)spinal.lib.bus.bmb.BmbParameter <and>
[error]   (addressWidth: Int,dataWidth: Int,sourceWidth: Int,contextWidth: Int,lengthWidth: Int,alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.Kind,alignmentMin: Int,accessLatencyMin: Int,canRead: Boolean,canWrite: Boolean,canExclusive: Boolean,maximumPendingTransaction: Int)spinal.lib.bus.bmb.BmbParameter
[error]  cannot be applied to (addressWidth: Int, dataWidth: Int, lengthWidth: Int, sourceWidth: Int, contextWidth: Int, canRead: Boolean, canWrite: Boolean, alignment: spinal.lib.bus.bmb.BmbParameter.BurstAlignement.LENGTH.type, maximumPendingTransactionPerId: Int)
[error]   def getBmbParameter(plugin : IBusSimplePlugin = null) = BmbParameter(
[error]                                                           ^
[error] four errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 13 s, completed Jan 29, 2021, 6:08:51 PM
tom@thinkcenter:~/projects/VexRiscv.test$ 
```

So the problem is that SpinalHDL 1.4.2 isn't backwards compatible with 1.4.0.

Dolu1990, 2021-01-30:
Hoo right, got it now.

Hmm, maybe the best is that i released SpinalHDL 1.4.3 / sync VexRiscv master/dev branch to be sure everything is updated.
Would that be fine ?

tomverbeure, 2021-01-30:
Yes, that’s fine!

Dolu1990, 2021-02-01:
Rawrrrrr
https://github.com/SpinalHDL/VexRiscv/commit/36b3cd918896c94c4e8a224d97c559ab6dbf3ec9#diff-5634c415cd8c8504fdb973a3ed092300b43c4b8fc1e184f7249eb29a55511f91L1

So master should now be happy with 1.4.3 ^^

closed 2021-02-01 by Dolu1990

tomverbeure, 2021-02-01:
Thanks! It's compiling fine now. 

One small remark: I don't see a v1.4.3 tag in the SpinalHDL repo.  Can you add that? I use those to rewind my local SpinalHDL repo when comparing different versions.

Tom

Dolu1990, 2021-02-02:
Done ^^