SpinalHDL/VexRiscv #165: Instruction-address-misaligned trap not working on branch to odd address

jorisvr, 2021-02-13:
It looks like the BranchPlugin does not correctly generate *instruction-address-misaligned* traps in all cases where I expect them, even when `catchAddressMisaligned=true`.

I'm using the current github master 36b3cd918896c94c4e8a224d97c559ab6dbf3ec9 and configured with:
```
  new IBusSimplePlugin(
    resetVector = 0x80000000l,
    cmdForkOnSecondStage = true,
    cmdForkPersistence = false,
    prediction = STATIC,
    catchAccessFault = false,
    compressedGen = false
  ),
...
  new BranchPlugin(
    earlyBranch = true,
    catchAddressMisaligned = true
  ),
```

I test by running the following code:
```
80000000 <_start>:
80000000:  00000297  auipc t0,0x0
80000004:  01028293  addi  t0,t0,16 # 80000010 <loop>
80000008:  00128293  addi  t0,t0,1
8000000c:  00028067  jr    t0

80000010 <loop>:
80000010:  00100513  li    a0,1
80000014:  ffdff06f  j     80000010 <loop>

80000020 <__trap_vector>:
80000020:  00200513  li    a0,2
80000024:  ffdff06f  j     80000020 <__trap_vector>
```

In this case I would expect the `jr` instruction to trap because the target address is odd. However, the code runs as if the least significant address bit is ignored and ends up spinning in `loop`.

When I change the 3rd instruction of the above program to use branch offset +2 instead of +1, I do in fact get a trap as expected.

Looking at the source of BranchPlugin.scala, it seems to me that the least significant bit of the target address is simply ignored.

This issue may be related to #37 (I can't tell for sure, sorry if this is a duplicate).

Off-topic, I would like to say that I'm very happy with the VexRiscv.
It is so configurable that it fits exactly in my project, and it works absolutely fine. I'm not ready to use SpinalHDL for my own designs, but that is no problem because the generated VexRiscv model interfaces very nicely with my VHDL code.
I also love the concept of the *simple pipelined memory bus*. It is all so easy and pleasant to work with. Thanks!

Dolu1990, 2021-02-14:
Hi,

Hooo yes i agree, it "should" create an exception. The thing is that it is a weird case of the RISC-V spec which say for JALR : 

> The indirect jump instruction JALR (jump and link register) uses the I-type encoding. The target
address is obtained by adding the sign-extended 12-bit I-immediate to the register rs1, then setting
the least-significant bit of the result to zero.

"then setting the least-significant bit of the result to zero" is the tricky thing :)

So to me, i think it follow the spec. but i agree it is weird XD

> Off-topic, I would like to say that I'm very happy with the VexRiscv.

Thanks :)

jorisvr, 2021-02-15:
Sorry, my mistake.

Thanks for explaining this part of the spec to me. I was actually hoping it would work this way, but I got confused by a different section of the spec. I see now that it is actually pretty clearly spelled out.

I'll close this issue then. Great work!

closed 2021-02-15 by jorisvr