SpinalHDL/VexRiscv #166: A null pointer access has been detected in the JVM.

siluks-plumerai, 2021-03-22:
Hi Dolu, we are using vexriscv and it works really well! Actually this is first time I encountered problem I cannot solve on my own. Maybe you could help. I modified Briey demo. Everything was ok until I added `axiLiteMaster0` top interface connected directly to `dbus` in crossbar. So nothing complicated, but fails with below error. If comment out all references to `axiLiteMaster0` and `axiLiteMaster0Ctrl`  passes again. So its somehow related to this additional master interface. To reproduce just copy Briey.txt attached into current master tree Briey.scala and run compilation:
[Briey.txt](https://github.com/SpinalHDL/VexRiscv/files/6180629/Briey.txt)

```
slawek@solo:~/git/github/VexRiscv$ sbt "runMain vexriscv.demo.Briey"
[info] Loading settings for project vexriscv-build from plugins.sbt ...
[info] Loading project definition from /home/slawek/git/github/VexRiscv/project
[info] Loading settings for project root from build.sbt ...
[info] Set current project to VexRiscv (in build file:/home/slawek/git/github/VexRiscv/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running (fork) vexriscv.demo.Briey 
[info] [Runtime] SpinalHDL v1.4.3    git head : adf552d8f500e7419fff395b7049228e4bc5de26
[info] [Runtime] JVM max memory : 7109.5MiB
[info] [Runtime] Current date : 2021.03.22 09:29:29
[info] [Progress] at 0.000 : Elaborate components
[info] PcManagerSimplePlugin is now useless
[info] ERROR !
[info] A null pointer access has been detected in the JVM.
[info] This could happen when in your SpinalHDL description, you access an signal which is only defined further.
[info] For instance :
[info]   val result = Bool
[info]   result := a ^ b  //a and b can't be accessed there because they are only defined one line below (Software rule of execution order)
[info]   val a,b = Bool
[info]           
[error] Exception in thread "main" java.lang.NullPointerException
[error]         at spinal.lib.bus.amba4.axi.Axi4SharedArbiter.<init>(Axi4Arbiter.scala:159)
[error]         at spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$29$$anon$7.<init>(Axi4Crossbar.scala:253)
[error]         at spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$29.apply(Axi4Crossbar.scala:252)
[error]         at spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$29.apply(Axi4Crossbar.scala:188)
[error]         at scala.collection.TraversableLike$WithFilter$$anonfun$map$2.apply(TraversableLike.scala:683)
[error]         at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at scala.collection.TraversableLike$WithFilter.map(TraversableLike.scala:682)
[error]         at spinal.lib.bus.amba4.axi.Axi4CrossbarFactory.build(Axi4Crossbar.scala:188)
[error]         at vexriscv.demo.Briey$$anon$3.<init>(Briey.scala:431)
[error]         at vexriscv.demo.Briey.<init>(Briey.scala:280)
[error]         at vexriscv.demo.Briey$$anonfun$main$1.apply(Briey.scala:466)
[error]         at vexriscv.demo.Briey$$anonfun$main$1.apply(Briey.scala:465)
[error]         at spinal.core.internals.PhaseCreateComponent.impl(Phase.scala:2166)
[error]         at spinal.core.internals.PhaseContext.doPhase(Phase.scala:180)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$117.apply(Phase.scala:2442)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$117.apply(Phase.scala:2440)
[error]         at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2440)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2379)
[error]         at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:31)
[error]         at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2379)
[error]         at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2349)
[error]         at spinal.core.Spinal$.apply(Spinal.scala:353)
[error]         at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:160)
[error]         at vexriscv.demo.Briey$.main(Briey.scala:465)
[error]         at vexriscv.demo.Briey.main(Briey.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 2 s, completed Mar 22, 2021 9:29:30 AM
slawek@solo:~/git/github/VexRiscv$ 
```

Dolu1990, 2021-03-22:
Hi,

You created a AXI4lite like bus from the regular AXI4 definition right ?
If yes, the interconnect assume there is a ID feature of the AXI is enabled. So i would say, set the Axi4Config.useId = true with idWidth=0

It should be fine then.
Tell me how it goes :)

siluks-plumerai, 2021-03-22:
Thanks a lot, that was fast! Setting useId fix the problem in source file provided.

Actually at begging I wanted to use AxiLIte4, but crossbar requires Axi4
```
[error] /home/slawek/git/github/VexRiscv/src/main/scala/vexriscv/demo/Briey.scala:376:26: type mismatch;
[error]  found   : (spinal.lib.bus.amba4.axilite.AxiLite4, (Long, scala.math.BigInt))
[error]  required: (spinal.lib.bus.amba4.axi.Axi4Bus, spinal.lib.bus.misc.SizeMapping)
[error]       axiLiteMaster0Ctrl -> (0x50000000L,   4 kB),
```

So i changed types (but not variable names) to Axi4 and encountered described problem. 

By the way. Is worth to play with AxiLite4 at all? 
- Should I use Axi4Shared internally, then cast to AxiLIte4 on i/o interface (if that possible)?
- Or better give up AxiLite4 at all and use Axi4 with burst, caches etc disabled?

Dolu1990, 2021-03-22:
> AxiLite4

Personnaly i never realy used it and instead used either APB3 for slave, either a simple AXI4 as master on which i drive things as if that was AxiLite4.

> Axi4Shared internally, then cast to AxiLIte4

I don't think there is such cast, i would say use directly Axi4Shared for you master instead.

> Or better give up AxiLite4 at all and use Axi4 with burst, caches etc disabled?

Right, i would go for that :)

siluks-plumerai, 2021-03-22:
Thank you for explanation

siluks-plumerai, 2021-04-08:
HI again @Dolu1990, one question regarding using AXI4 master with AXI4Lite external slave. Can I adjust .scala to use only burst len = 0? When I simply use `useLen = false` in *Axi4Config* connected to crossbar I get following error. Maybe you have some recommended solution to this?

```
[error] Exception in thread "main" spinal.core.SpinalExit: 
[error]  
[error] ********************************************************************************
[error] ********************************************************************************
[error] (toplevel/??? :  UInt[8 bits]) can't drive toplevel/[Axi4SharedArbiter]/io_sharedInputs_0_arw : Stream[Axi4Arw] because this last one doesn't has the corresponding pin
[error]     spinal.lib.bus.amba4.axi.Axi4Priv$.driveWeak(Axi4Channel.scala:226)
[error]     spinal.lib.bus.amba4.axi.Axi4Priv$.driveAx(Axi4Channel.scala:239)
[error]     spinal.lib.bus.amba4.axi.Axi4Arw$StreamPimper.drive(Axi4Channel.scala:279)
[error]     spinal.lib.bus.amba4.axi.Axi4Shared.$greater$greater(Axi4Shared.scala:20)
[error]     spinal.lib.bus.amba4.axi.Axi4Shared.$less$less(Axi4Shared.scala:18)
[error]     spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$29$$anon$7$$anonfun$44.apply(Axi4Crossbar.scala:269)
[error]     spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$29$$anon$7$$anonfun$44.apply(Axi4Crossbar.scala:268)
[error]     spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$29$$anon$7.<init>(Axi4Crossbar.scala:268)
[error]     spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$29.apply(Axi4Crossbar.scala:252)
[error]     spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$29.apply(Axi4Crossbar.scala:188)
[error]     spinal.lib.bus.amba4.axi.Axi4CrossbarFactory.build(Axi4Crossbar.scala:188)
[error]     vexriscv.demo.Briey$$anon$3.<init>(Briey.scala:452)
[error]     vexriscv.demo.Briey.<init>(Briey.scala:291)
[error]     vexriscv.demo.Briey$$anonfun$main$1.apply(Briey.scala:493)
[error]     vexriscv.demo.Briey$$anonfun$main$1.apply(Briey.scala:492)
[error]     vexriscv.demo.Briey$.main(Briey.scala:492)
[error]     vexriscv.demo.Briey.main(Briey.scala)
```

siluks-plumerai, 2021-04-09:
Ive just realized that AXI master uses burst = 8 even when software reads only one data word

It just the same burst that is used by cpu dbus I guess.

So the only way to get burst=1 with axi master is to do axi->apb->axi conversions @Dolu1990 ?

Dolu1990, 2021-04-12:
I would say, your bridge from AxiLite to Axi should put the burst feature on the bus, and force it to 0 all the time. That way the Axi4SharedArbiter, which seems to do the assumption that the burst signal is used, will not crash

Dolu1990, 2021-04-12:
Same for the "last" pin feature, force it to 1 in the bridge

siluks-plumerai, 2021-04-13:
I decided to use external axi to axi lite adapter  https://github.com/alexforencich/verilog-axi/blob/master/rtl/axi_axil_adapter.v works well with vexriscv 

but I will do more experiments with doing this inside scala

closed 2021-04-13 by siluks-plumerai