SpinalHDL/VexRiscv #168: Debugin Problems with compressed instructions

BLangOS, 2021-03-27:
I am evaluating the Debug Interface with a VHDL testbench. Everything works fine when I use the 32 Bit instruction encoding.
But when I used the compressed instruction set I get unexpected results when the instructions (compressed or not) are not word aligned.
Can the Debug Interface even used with compressed instructions?

Dolu1990, 2021-03-27:
> Can the Debug Interface even used with compressed instructions?

It can't, basicaly, the decompressor is at the end of the fetch stage, while the instruction injector is in the front of the decode stage

So, this is the expected behaviour

BLangOS, 2021-03-27:
Sorry, I did not ask my question precisely enough. Therefore I would like to ask again.
I stopped the processor via debug interface and then of course only injected 32-bit instructions to the debug interface. However, the interrupted program also contains compressed instructions; it was compiled using gcc with '-march = rv32ic'.
If the program was stopped accidentally on instructions with non-aligned word addresses, the processor crashed when I tried to continue the interrupted program via Step or Contiunue.
So my more precise question is whether debugging is possible with programs that have been compiled with the ISA extension '-march = rv32ic'. I suspect that doesn't work at the moment.

Dolu1990, 2021-03-29:
Hmm debugging with RVC isn't extensively tested, but should normaly work. What CPU configuration / SoC do you use ?

BLangOS, 2021-03-30:
Below you find I system that put together to make the error visible.

The system has the following address layout: 
ROM: 0x80000000-0x80003FFF
RAM: 0x00000000-0x00000FFF (not used in the testbench)
PeripheralRegister: 0x00020000

The process called "ROM_RAM_and_Peripherals" describes ROM, RAM and a peripheral register. It cannot be synthesized. 
Two test programs with the same functionality are available in the testbench. The difference is that in Program 2 most of the instructions are word-aligned, but in Program 1 they are not.

Both programs first load register t0 with 0x00020000 (address of "PeripheralRegister"), t1 with 1 and t2 with 0. Register t2 is then incremented 8 times in a loop. Then its value is written to the "PeripheralRegister" and thus made visible to the outside. This means that the values 0x00000008, 0x00000010, 0x00000018, ... must appear one after the other in the "PeripheralRegister". 

The process, called "Debugging" serves the debug interface. After a delay, it halts the processor, then lets it take a step, and then lets the program resume. The delay is chosen so that the processor is halted during the Increment instructions which are not word-aligned in Program 1.
If Program 1 is selected, it can be seen that a bus error occurs on the instruction bus after the resume and the subsequent value of the "PeripheralRegister" is incorrect. Instead of 0x00000020, the value 0x0000001f appears in "PeripheralRegister" after the resume. Program 2, on the other hand, is processed without errors. 
The timing below shows a simulation run with Program 1: 
[image]

If the call of the "Step" function is removed in the "Debugging"-process, there is no error with either program. The program then always continues correctly after the resume. 

The processor is a modified version of the VexRiscvAhbLite3, below I enclose its scala description "VexRiscvAhbLite3_bl.scala" and the generated "VexRisc.vhd". Further you will find my testbench "VexRiscv_AhbLite3_dbg_tb.vhd" and a script "VexRiscv_AhbLite3_dbg.do" for the ModelSim simulator (I use the Intel Edition).

Hope this will help.

[Debugging_Problem_Compressed.zip](https://github.com/SpinalHDL/VexRiscv/files/6227983/Debugging_Problem_Compressed.zip)

BLangOS, 2021-03-31:
Thank you very much for your very helpful explanations.

I do only partially understand the details of your explanation (at the moment I'm still a newbie to SpinalHDL) but I interpreted it in such a way that the decompressor's buffer is invalidated when a step command is issued. A subsequent jump to the current PC (which seems to be unnecessary from the programs point of view) then refills this buffer if necessary.

I intensively studied the source of the VexRiscv OpenOCD target code ([vexriscv.c](https://github.com/SpinalHDL/openocd_riscv/blob/riscv_spinal/src/target/vexriscv.c)) but could not find this requirement modelled in the code. Maybe this code currently cannot support compressed instructions, or maybe I overlooked the associated code lines.
Nervertheless I plan to use an own GDBServer and not OpenOCD for debugging with GDB and your explanations will help me to support compressed program code.

Due to your tricky hints I added two commands to the testbench after the step command. The first command reads the PC and the second command writes the just read value back to the PC. When writing the PC precisely a jump command is executed by means of injection. And now it works :-).

Below you find the modified testbench and here is the new timing diagram, which now shows the expected sequence:
[image]

I have taken note of your further explanations on systems with MMU and/or the DBusCachedPlugin with great interest, as soon as I find the time I will take a closer look.

Thank you again very much for your very helpful hints.

[Debugging_Compressed.zip](https://github.com/SpinalHDL/VexRiscv/files/6235628/Debugging_Compressed.zip)

closed 2021-03-31 by BLangOS

Dolu1990, 2021-03-31:
> I intensively studied the source of the VexRiscv OpenOCD target code (vexriscv.c) but could not find this requirement modelled in the code.

I think it is likely done by : 
https://github.com/SpinalHDL/openocd_riscv/blob/riscv_spinal/src/target/vexriscv.c#L742

> The first command reads the PC and the second command writes the just read value back to the PC. When writing the PC precisely a jump command is executed by means of injection. And now it works :-).

It may even be possible to just do a JAL relatif to the current PC, that way only one instruction is required to "reload" the pipeline.

But i think the best is to stay general and not try to trick things too much ^^ 

have fun :)

BLangOS, 2021-03-31:
Oh yes, now I see it. The PC is written when its image is valid, even if it is not dirty. Very good hidden.
Thanks :-)

BLangOS, 2021-03-31:
I tested `JAL x0,0` and it also works. Yes, thats much easier.
:-)