SpinalHDL/VexRiscv #169: Questions . How much the max frequency we can get for EC5G.

akshar001, 2021-03-29:
Hello,

I am totally new to this thing and want to learn most of this thing and have found your project amazing.

My question is what is the max clock we can get for SOCs if we use EC5G family of lattice.

It might be very basic question but thought of asking it here.

Dolu1990, 2021-03-29:
Hi,

Basicaly i do not have a flow for ECP5 FPGA, but i gess it is around the speed of a Cyclone V ?
I'm not sure.

akshar001, 2021-03-29:
Can we runt the cpu core at 1Ghz anyhow?

Dolu1990, 2021-03-29:
Only in simulation  ^^

On cheap FPGA you can run linux at 100 Mhz, probably with expensive FPGA you can go to 300-400 Mhz

akshar001, 2021-03-29:
What if we use external pll system and supply it to inner fpga system clock?

Dolu1990, 2021-03-29:
You will just make the logic go to fast and corrupt

akshar001, 2021-03-29:
Ohkay. So there is no way we can make them run faster. What's the bottleneck here?

Dolu1990, 2021-03-29:
CPU stall logic

akshar001, 2021-03-29:
Is there any other way we can make this faster?

Dolu1990, 2021-03-29:
cryo cooling ^^

akshar001, 2021-03-29:
haha, so ultimate way to make this things faster is to make our own fpga!

akshar001, 2021-03-29:
what we can do to avoid cpu stall logic?
Any pipeline related thing?

Dolu1990, 2021-03-29:
you can't realy avoid that in a pipelined CPU. things from down in the stge have to propagate up

akshar001, 2021-03-29:
XC7A35T-2CPG236C

This says it can run up to 450Mhz. Can we really do it?

Dolu1990, 2021-03-29:
The 450Mhz advertised is about the clock tree, as soon you chain logic it will fall down realy fast

Dolu1990, 2021-03-29:
that CPU in polarfire isn't part of the FPGA stuff, it's directly engraved in the silicon. that's why it is that fast

akshar001, 2021-03-29:
Yes saw that. there are many more eFPGA concepts are coming

akshar001, 2021-03-29:
I want to learn all this basic concepts and learn spinalhdl , can you guide me some way?

akshar001, 2021-03-29:
I want to make one open source cheap fpga board with quad core vexRISC-V , 1GB RAM,HDMI,USB,WIFI,ETHERNET everything. like in under 100$ i want to sell it! Would it be cool?

closed 2021-03-30 by akshar001

akshar001, 2021-03-31:
Okay! Wow! Let's get on with this!

akshar001, 2021-03-31:
I will need to give you 2-3 board at-least for testing!

Dolu1990, 2021-03-31:
> Hmm im working on having a USB host controller for linux <3

I mean that i'm already working on USB host for artyA7 and Ulx3s, could also be used for other boards

> I will need to give you 2-3 board at-least for testing!

I personnaly try to keep as few board as possible and avoid spliting my times on too many plateforms (as that's endless) XD

Please, if you bring a new board,  you have to port the SoC for it. I can help with informations.

akshar001, 2021-03-31:
Okay. I might be going to use ECP5  - 85F.
Later if we need to port your code, i will certainly do it.

akshar001, 2021-03-31:
So we will have 85F, DDR3 RAM 1 GB , RGMII ethernet , USB , UART , uSD CARD , WiFi/BLE in general in this board. 
Plus flexible interface of making sub-modules , like USB expander o UART expander. I will be manufacturing this in India , so i can get most of the cost to very cheaper level , I want this board to be a super cheap so that so many people can use it.

akshar001, 2021-04-13:
@Dolu1990 have you ever used XC7K325T-2FFG900i

Dolu1990, 2021-04-13:
No, never used FPGA outside the low cost range.

akshar001, 2021-04-13:
I have purchased 2 of them. Let's see if i can make any board out of them. If we be able to make 400$ FPGA dev kit using this chip it would be good?

Dolu1990, 2021-04-13:
400 $ is realy much, also, is vivado free to use for such big FPGA ?
One thing (i think) is that i don't realy see a good usage of FPGA above 100K lut ^^ 
I mean, with what do you fill such FPGA ? (sure you can fill it with non-softcore-designed CPU)

akshar001, 2021-04-13:
Understood