SpinalHDL/VexRiscv #170: Linefill Question

openpowerwtf, 2021-03-30:
I'm starting to simulate a modified version of the core with caches enabled, and Litex-gen'd Wishbone/memory.  Instruction fetching works fine and some stores have completed, but the first data load hangs.  The IC/DC line loaders are operating differently (and logic is different because of stores/uncacheables).  The IC line loader goes valid at the start of the load sequence, but the DC line loader waits for all 8 commands to be sent before going valid; it only sees the last valid response, so the fill never completes.   

Seems like I must have something configured incorrectly (or am completely confused 😕) .  Any ideas?

Dolu1990, 2021-03-30:
Hi,

Both cache have a different refill logic, both cache reschedule the instruction if there was a miss (refill then retry)

Hmm in your case, the data cache hang ?
That's weird, do you have a wave to share ?

openpowerwtf, 2021-03-30:
I know, this must be something basic.  This is my first FPGA-target/Wishbone attempt - I have gen'd bitstreams and blinked LEDs through CSR (code is in on-board 'ROM').   But before this, I have been doing simulation with the core+AXI (software AXI model).

_Unfortunately_ I'm doing some things non-VexRiscv-like 🤣:  the core doesn't run Risc V, its hacked into the Litex top-level, and I'm using Vivado for sim.  I broke a lot of stuff along the way but I don't think I touched much in the caches.

I think Vivado is supposed to be able to save to vcd.   I will attach a clip of a few signals in case there's something obvious (read @2000, which is uninitialized BRAM).   

[image]

GrobiG, 2021-05-01:
I have experienced the same issue a few month back, I used the avalon example, I just needed a simple implementation no linux, mmu stuff, with the cached instructionfetcher and databus. 
The non-cached versions wouldn't work at all. 
The instructionfetcher worked fine reading and the databus worked in most cases fine writing but reading on the databus always returned a zero or bogus value. 
I currently have no simulation screenshots available as I abandoned the project after two weeks of unsuccessfull trial and error and started to create my own risc-v soc designed especially to work with the avavalon fabric but it is lightyears away from what vexrisc-v could do but as vexrisc-v just does not work for me I basically had no ohther choice.

Dolu1990, 2021-05-01:
When you have issues with the bus you can send me the wave, i can easily check stuff and fix if i have access to a proper wave. From that point it is easy to figure out on which side stuff is turning bad.

GrobiG, 2021-05-30:
I found some time to re-create my test project and I am starting to think there is nothing wrong with the cache implementation, I am just setting it up wrong and I am making a big mistake with the static memory translator plugin.

In my setup I put "devices/pios" starting at address 0x0201 0000 and for those the databus cache should be bypassed, in the .scala file I have configured it like this:
```
          new DBusCachedPlugin(
            config = new DataCacheConfig(
              cacheSize         = 4096,
              bytePerLine       = 32,
              wayCount          = 1,
              addressWidth      = 32,
              cpuDataWidth      = 32,
              memDataWidth      = 32,
              catchAccessError  = true,
              catchIllegal      = true,
              catchUnaligned    = true
            ),
            memoryTranslatorPortConfig = null
          ),
          new StaticMemoryTranslatorPlugin(
            ioRange      = _(31 downto 25) === 0xFE
          ),
```
and that seems to be wrong.

In my program I am constantly reading from 0x02010040 and writing the returned value to 0x02010050, from what I can see in the simulation the databus_avalon interfaces read signal only gets asserted once and not on the following reads from 0x02010040 so I guess the other times it reads from the cache, which probably never again gets updated as I never write anything to that address.

What is the correct usage of the StaticMemoryTranslatorPlugin?

Dolu1990, 2021-05-31:
So, you want your uncached access to be at  0x02xx_xxxx right ?
Then you need  ioRange      = _(31 downto 25) === 0x02

Right ?

GrobiG, 2021-05-31:
Yes, correct, I would like everything starting from address 0x02010000 to be "uncached", I changed the config and now it works totally fine. Thank you a lot for that!!!

But is there any more documentation available regarding this except for: 
"Static memory translator plugin which allows to specify which range of the memory addresses is I/O mapped and shouldn't be cached."
and every example which uses it configured it like "ioRange      = _(31 downto 28) === 0xF" which makes it kinda hard to guess how it works.

openpowerwtf, 2021-08-21:
I have created the same fail using an unmodified core, so I must be missing something in my process.  I did use this version as a guide when creating mine, so it's not surprising.  I still need to try a core like the one used for Linux-on-Litex.  

I built:

```
sbt "runMain vexriscv.demo.VexRiscvCachedWishboneForSim"
```
and then ran a simple verilator sim, returning 'lw a0, 0(a0)' for instruction fetches.  I get 8 WB I reads (00:1C) and 8 WB D reads (00:1C), and loader_valid is stuck active, since the ack's are already consumed by the commands being sent.

Dolu1990, 2021-08-23:
@GrobiG 
> But is there any more documentation available regarding this except for:

The doc is a bit lacking XD

> ...man I am such an idiot, I am sorry I have wasted your time with this.

No worries, the paradigme where you can give a landa function to a CPU parameter to specify how to implement address filtering isn't common :)

@openpowerwtf
Do you have a wave ? (VCD or FST) ?

openpowerwtf, 2021-08-24:
Yes, thanks for taking a look.  Every time I get back to looking at this, it seems like the logic is working as expected (obviously not!).
[
[gtkwave.zip](https://github.com/SpinalHDL/VexRiscv/files/7038773/gtkwave.zip)
](url)

Dolu1990, 2021-08-24:
@openpowerwtf Ahhh, the issue you have in the testbench is that you are providing the dbus_rsp before/durring consuming  the dbus_cmd transaction.

For it to work, you need to have dBusCmdMasterPipe turned on. I can just see now that the wishbone example in the repo do not turn it on XD (i just pushed a fix https://github.com/SpinalHDL/VexRiscv/commit/5c7e4a0294c785b58b424e49b4696855143370cb)

Did you had dBusCmdMasterPipe turned on in the DBusCachedPlugin ?

openpowerwtf, 2021-11-09:
Closing - running Litex BIOS and external RAM on cmod-A7 with PPC version of VexRiscv.

closed 2021-11-09 by openpowerwtf