SpinalHDL/VexRiscv #172: How to access memory of CPU core?

GuitarZeiZei, 2021-04-07:
Hello,

I'm extending a coprocessor instruction and defining a coprocessor interface like ROCC, but I don't know how to have access to the memory and read some data. Should I connect to the memory bus or something eles? I wonder if there is any example. 

Thanks.

Dolu1990, 2021-04-07:
Hi.

Currently the data cache is tightly coupled with the CPU pipeline. So, you can't have a interface to the L1 D$ as ROCC. 

But meanwhile, i recently added the CPU support to Vex using a coprocessor like interface : 
https://github.com/SpinalHDL/VexRiscv/tree/dev#fpu

Basicaly, the coprocessor instructions can ask the data cache to encode new load/store instruction (which can even be 64 bits data transfers). See https://github.com/SpinalHDL/VexRiscv/blob/fpu/src/main/scala/vexriscv/plugin/FpuPlugin.scala#L158

Those load/store have to follow the regular load store encoding format. Which is the case for the RISC-V FPU :D

So i would say, check out https://github.com/SpinalHDL/VexRiscv/blob/fpu/src/main/scala/vexriscv/plugin/FpuPlugin.scala#L219 in depth to see if that can fit you needs, and if adapdtations are possible.

GuitarZeiZei, 2021-04-08:
OK, I'll check and see.

Thank you very much for your suggestion. : )