SpinalHDL/VexRiscv #175: Briey SoC (with SpinalHDL version > 1.4.0) does not work anymore on Terasic De0-nano (Cyclone 4)

amitkulkarnis, 2021-04-14:
Hi,

I updated the Briey SoC by generating it with the current toolchain that uses SpinalHDL 1.4.3. I used the updated Briey.v in the FPGA project that is present on Google drive. 
I observe that access to SDRAM is prohibited and the content of the SDRAM (after flashing the application elf file) still shows 0xfffffff. 
Can you please check if there is bug?
Thank you.

Dolu1990, 2021-04-15:
Hi
Does the simulation of Briey works for you ? (as described in the main readme) I tested that simulation not so long ago, it was fine for me.

I have to say i haven't use Briey since years.

What do you mean by " access to SDRAM is prohibited " ?

amitkulkarnis, 2021-04-16:
Hi,
Briey SoC on the current version of the toolchain works fine on simulation (VGA example).
However, when we use it with the Quartus project (https://drive.google.com/drive/folders/0B-CqLXDTaMbKZGdJZlZ5THAxRTQ?usp=sharing) by replacing the Briey.v (generated using current toolchain) we have the problem.
The problem is: 
When I dump the *.elf file into the SDRAM (using riscv gdb), the contents of the SDRAM is always 0xffffffff. Hence the core does not respond when I try to run an application such as UART. This is what "access to SDRAM" means, the gdb cannot dump the application. The Briey.v (generated with the current toolchain) is not suitable with the Quartus project that you have provided.
Thanks.

Dolu1990, 2021-04-17:
Question : 
Does Briey boot by himself ? i mean, without using the JTAG, are things fine ?
Does the io pin names still match between the old briey and the new briey ?

amitkulkarnis, 2021-04-19:
Hi,

1) No SoC does not respond well with/without JTAG.
2) There is one major change in the Briey SoC's IO:
old briey: "  output  io_sdram_DQ_writeEnable,"
new briey:   "output     [15:0]   io_sdram_DQ_writeEnable,"

Thanks.

Dolu1990, 2021-04-20:
Cool :D
> It would be great if you adopt this change in Scala source code for Briey.

Hmm i think i did io_sdram_DQ_writeEnable with one bit for each pin to help inferring the tristate enable register into the io primitive, so i would say the best would be to update the online projects to follow this new way of doing things. Does that seem fair ?

amitkulkarnis, 2021-04-20:
Yes, that would do. 
I am attaching the top-level VHDL file to fix the problem.
[DE0_NANO.zip](https://github.com/SpinalHDL/VexRiscv/files/6345790/DE0_NANO.zip)

closed 2021-04-20 by amitkulkarnis

Dolu1990, 2021-04-20:
Thanks, i updated the de0_nano zip online :)