SpinalHDL/VexRiscv #178: WIDTH MISMATCH Error when using UserInterruptPlugin with some configuration options

BLangOS, 2021-05-03:
I tried to use User Interrupts as explained in ["GenCustomInterrupt.scala"](https://github.com/SpinalHDL/VexRiscv/blob/dev/src/main/scala/vexriscv/demo/GenCustomInterrupt.scala).
With some configuration options (e.g. `catchIllegalInstruction = true` in `DecoderSimplePlugin` or `catchAddressMisaligned = true` in `BranchPlugin`) I got an error when generating VHDL-Code:
`WIDTH MISMATCH on (toplevel/CsrPlugin_trapCause :  UInt[5 bits]) := (toplevel/CsrPlugin_exceptionPortCtrl_exceptionContext_code :  UInt[4 bits]) at`
`    vexriscv.plugin.CsrPlugin$$anon$15$$anonfun$12.apply$mcV$sp(CsrPlugin.scala:942)` ...

As far as I understand the error message the code in [CsrPlugin.scala:942](https://github.com/SpinalHDL/VexRiscv/blob/75bbb28ef62636dd0d4d3741c6e559a911fc85af/src/main/scala/vexriscv/plugin/CsrPlugin.scala#L942) fails due to different vector length on the left and right side of the assignment.
Changing the line to `trapCause := exceptionPortCtrl.exceptionContext.code.resized` did fix the error message. But is this a correct solution?

Regards, Bernhard.

Dolu1990, 2021-05-04:
Thanks for the report ^^

This should now be fixed via the above fix.

Dolu1990, 2021-05-04:
> Changing the line to trapCause := exceptionPortCtrl.exceptionContext.code.resized did fix the error message. But is this a correct solution?

I don't think it would have been, because in the reversed situation (bigger exception width) it would lose some bits :)

commit e78c0546 (2021-05-04): fix #178
closed 2021-05-04 by Dolu1990