SpinalHDL/VexRiscv #180: Regression Test not all pass

EggerStephan, 2021-05-05:
Hello everyone,
i have a problem with the regression test
Here is my configuration

```
verilator --version
Verilator 4.202 2021-04-24 rev v4.202-4-g422c076f
```

```g++ --version
g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I build the GenFull demo with sbt

 ```
sbt "runMain vexriscv.demo.GenFull"
[info] Loading settings for project vexriscv-build from plugins.sbt ...
[info] Loading project definition from /home/eggerstephan/VexRiscvCoreProject/VexRiscv/project
[info] Updating ProjectRef(uri("file:/home/eggerstephan/VexRiscvCoreProject/VexRiscv/project/"), "vexriscv-build")...
[info] Done updating.
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[info] Loading settings for project root from build.sbt ...
[info] Set current project to VexRiscv (in build file:/home/eggerstephan/VexRiscvCoreProject/VexRiscv/)
[info] Updating ...
[info] Done updating.
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[info] Compiling 85 Scala sources to /home/eggerstephan/VexRiscvCoreProject/VexRiscv/target/scala-2.11/classes ...
[warn] /home/eggerstephan/VexRiscvCoreProject/VexRiscv/src/main/scala/vexriscv/demo/smp/VexRiscvSmpLitexMpCluster.scala:1:1:
[warn] Found names but no class, trait or object is defined in the compilation unit.
[warn] The incremental compiler cannot record the dependency information in such case.
[warn] Some errors like unused import referring to a non-existent class might not be reported.
[warn]
[warn] package vexriscv.demo.smp
[warn] ^
[warn] there were 35 deprecation warnings; re-run with -deprecation for details
[warn] there were two feature warnings; re-run with -feature for details
[warn] three warnings found
[info] Done compiling.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.protobuf.UnsafeUtil (file:/home/eggerstephan/.sbt/boot/scala-2.12.7/org.scala-sbt/sbt/1.2.7/protobuf-java-3.3.1.jar) to field java.nio.Buffer.address
WARNING: Please consider reporting this to the maintainers of com.google.protobuf.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Packaging /home/eggerstephan/VexRiscvCoreProject/VexRiscv/target/scala-2.11/vexriscv_2.11-2.0.0.jar ...
[info] Done packaging.
[info] Running (fork) vexriscv.demo.GenFull
[info] [Runtime] SpinalHDL v1.4.3    git head : adf552d8f500e7419fff395b7049228e4bc5de26
[info] [Runtime] JVM max memory : 1554.0MiB
[info] [Runtime] Current date : 2021.05.05 14:07:58
[info] [Progress] at 0.000 : Elaborate components
[info] [Progress] at 1.123 : Checks and transforms
[info] [Progress] at 1.621 : Generate Verilog
[info] [Warning] 201 signals were pruned. You can call printPruned on the backend report to get more informations.
[info] [Done] at 2.065
[success] Total time: 57 s, completed May 5, 2021, 2:08:00 PM
```

After that i start the regression test 

```
 cd src/test/cpp/regression/
~/VexRiscvCoreProject/VexRiscv/src/test/cpp/regression$ make clean run
```
```
rm -rf obj_dir
cp ../../../../VexRiscv.v*.bin . | true
cp: cannot stat '../../../../VexRiscv.v*.bin': No such file or directory
verilator -cc  ../../../../VexRiscv.v  -O3 -CFLAGS -std=c++11 -LDFLAGS -pthread  -CFLAGS -DREGRESSION_PATH='\".//\"' -CFLAGS -DIBUS_CACHED -CFLAGS -DIBUS_DATA_WIDTH=32 -CFLAGS -DDBUS_DATA_WIDTH=32 -CFLAGS -DDBUS_CACHED -CFLAGS -DREDO=10 -CFLAGS -pthread -CFLAGS -Wno-unused-result -CFLAGS -DTHREAD_COUNT=8 -CFLAGS -O3  -O3 -CFLAGS -DTIMER_INTERRUPT -CFLAGS -DEXTERNAL_INTERRUPT -CFLAGS -DDHRYSTONE -CFLAGS -DSTALL=1 -CFLAGS -DCSR -CFLAGS -DISA_TEST -CFLAGS -DMMU -CFLAGS -DMUL -CFLAGS -DDIV -CFLAGS -DDEBUG_PLUGIN -CFLAGS -DDEBUG_PLUGIN_STD -CFLAGS -DTRACE_START=0 --gdbbt  -Wno-UNOPTFLAT -Wno-WIDTH --x-assign unique --exe main.cpp
No stack.
make  -j8 -C obj_dir/ -f VVexRiscv.mk VVexRiscv
make[1]: Entering directory '/home/eggerstephan/VexRiscvCoreProject/VexRiscv/src/test/cpp/regression/obj_dir'
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DREGRESSION_PATH=\".//\" -DIBUS_CACHED -DIBUS_DATA_WIDTH=32 -DDBUS_DATA_WIDTH=32 -DDBUS_CACHED -DREDO=10 -pthread -Wno-unused-result -DTHREAD_COUNT=8 -O3 -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DDHRYSTONE -DSTALL=1 -DCSR -DISA_TEST -DMMU -DMUL -DDIV -DDEBUG_PLUGIN -DDEBUG_PLUGIN_STD -DTRACE_START=0  -std=gnu++14 -Os -c -o main.o ../main.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DREGRESSION_PATH=\".//\" -DIBUS_CACHED -DIBUS_DATA_WIDTH=32 -DDBUS_DATA_WIDTH=32 -DDBUS_CACHED -DREDO=10 -pthread -Wno-unused-result -DTHREAD_COUNT=8 -O3 -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DDHRYSTONE -DSTALL=1 -DCSR -DISA_TEST -DMMU -DMUL -DDIV -DDEBUG_PLUGIN -DDEBUG_PLUGIN_STD -DTRACE_START=0  -std=gnu++14 -Os -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DREGRESSION_PATH=\".//\" -DIBUS_CACHED -DIBUS_DATA_WIDTH=32 -DDBUS_DATA_WIDTH=32 -DDBUS_CACHED -DREDO=10 -pthread -Wno-unused-result -DTHREAD_COUNT=8 -O3 -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DDHRYSTONE -DSTALL=1 -DCSR -DISA_TEST -DMMU -DMUL -DDIV -DDEBUG_PLUGIN -DDEBUG_PLUGIN_STD -DTRACE_START=0  -std=gnu++14 -Os -c -o verilated_dpi.o /usr/local/share/verilator/include/verilated_dpi.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VVexRiscv.cpp VVexRiscv_VexRiscv.cpp VVexRiscv__Dpi.cpp VVexRiscv__Slow.cpp VVexRiscv_VexRiscv__Slow.cpp VVexRiscv__Syms.cpp > VVexRiscv__ALL.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DREGRESSION_PATH=\".//\" -DIBUS_CACHED -DIBUS_DATA_WIDTH=32 -DDBUS_DATA_WIDTH=32 -DDBUS_CACHED -DREDO=10 -pthread -Wno-unused-result -DTHREAD_COUNT=8 -O3 -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DDHRYSTONE -DSTALL=1 -DCSR -DISA_TEST -DMMU -DMUL -DDIV -DDEBUG_PLUGIN -DDEBUG_PLUGIN_STD -DTRACE_START=0  -std=gnu++14 -Os -c -o VVexRiscv__ALL.o VVexRiscv__ALL.cpp
Archive ar -rcs VVexRiscv__ALL.a VVexRiscv__ALL.o
g++    main.o verilated.o verilated_dpi.o VVexRiscv__ALL.a   -pthread    -o VVexRiscv
make[1]: Leaving directory '/home/eggerstephan/VexRiscvCoreProject/VexRiscv/src/test/cpp/regression/obj_dir'
./obj_dir/VVexRiscv
BOOT
```
Then the test starts and get this message at the end
```
****************************************************************
Had simulate 2505307 clock cycles in 4.11657 s (608.591 Khz)
REGRESSION FAILURE 110/1094
****************************************************************
```

The errors are all the same
```
regFile write missmatch :
 REF: RF[1] = 0xffffffff
 DUT: RF[1] = 0x0
FAIL I-CSRRC-01 at PC=80000024 REF PC=80000024 REF I=3400b0f3 time=372
```
Am i missing a setting or something?
Thanks,

Stephan

Dolu1990, 2021-05-05:
Gen full isn't intended to pass all regression tests.

Normaly, it should pass all of  : 
make clean run REDO=10 CSR=no MMU=no 
Can you give a try ?

EggerStephan, 2021-05-06:
Thanks for the fast answer.
With this settings all tests passes

thanks for the help

closed 2021-05-07 by EggerStephan