SpinalHDL/VexRiscv #183: Arduino with Murax.

akshar001, 2021-05-12:
Hello,

I am developing one board which will eventually support Murax SOC with ICE40,
My goal is to develop an Open source MCU with Open source Hardware and software, basically everything from top to scratch is open source.

I am willing to develop an Arduino integration with Murax SOC as well as other blocks.
The help i need from you is how can i change blocks easily. My end goal is to have a user flexibility in their controller for their supposive interfaces. Someone wants to have 3 channel UART or 2 SPI or 2i2C or just use GPIOs. They can easily reconfigure the SOC and upload an Arduino code into it.

I think this is a most amazing thing Arduino can support right now!

Any suggestions how can i easily develop a framework where anyone can change blocks easily would be much helpful.

Dolu1990, 2021-05-17:
Hi,

So, i would say the easy way would be to create a new scala main, which take arguments to drive the generation of more peripherals and generate some proper board support packages (soc.h, ...) . I haven't done this kind of things soo much so far, but it should work just fine.

Else there is SaxonSoC, but my guess is that Murax is simpled to start with :)

akshar001, 2021-05-24:
Okay, i will use murax, it's perfect for iCE40 right?

Dolu1990, 2021-05-24:
Sure, murax is good for ice40 ^^

closed 2021-05-26 by akshar001