SpinalHDL/VexRiscv #186: Building MuraxSOC fails

haritshukla, 2021-06-10:
Hello,
We are trying to build MuraxSOC for Arty-35,

It's failing 

sbt "runMain vexriscv.demo.MuraxWithRamInit" 

←[0m[←[0m←[31merror←[0m] ←[0m←[0m       at xsbt.boot.Launch$.apply(Launch.scala:44)←[0m
←[0m[←[0m←[31merror←[0m] ←[0m←[0m       at xsbt.boot.Launch$.apply(Launch.scala:21)←[0m
←[0m[←[0m←[31merror←[0m] ←[0m←[0m       at xsbt.boot.Boot$.runImpl(Boot.scala:78)←[0m
←[0m[←[0m←[31merror←[0m] ←[0m←[0m       at xsbt.boot.Boot$.run(Boot.scala:73)←[0m
←[0m[←[0m←[31merror←[0m] ←[0m←[0m       at xsbt.boot.Boot$.main(Boot.scala:21)←[0m
←[0m[←[0m←[31merror←[0m] ←[0m←[0m       at xsbt.boot.Boot.main(Boot.scala)←[0m
←[0m[←[0m←[31merror←[0m] ←[0m←[0mjava.io.IOError: java.lang.RuntimeException: /packages cannot be represented as URI←[0m

akshar001, 2021-06-10:
Issue was due to JAVA version , this project requires specific 1.2.7 sbt and JDK 8 or JDK 11

closed 2021-06-10 by haritshukla

Dolu1990, 2021-06-11:
Which java version did you had ?

akshar001, 2021-06-11:
We had 16.0 and sbt 1.5

akshar001, 2021-06-11:
We generated muraxSOC for Arty-35 and also synthesised it in vivado it all worked fine

Dolu1990, 2021-06-11:
I didn't even knew there was a java 16.0 XD

> We generated muraxSOC for Arty-35 and also synthesised it in vivado it all worked fine

Cool :)

akshar001, 2021-06-11:
Where is a scala code for murax?
We tried to find it in this repo but we didn't find it eventually

Dolu1990, 2021-06-11:
Should be there : 
https://github.com/SpinalHDL/VexRiscv/blob/75bbb28ef62636dd0d4d3741c6e559a911fc85af/src/main/scala/vexriscv/demo/Murax.scala#L155

dwivedi998, 2021-07-16:
I want to download version 1.2.8 of SBT and I have downloaded a zip file, but I am unable find any installation procedure after that, Can anybody tell me what to do.

Dolu1990, 2021-07-20:
@dwivedi998 
Hi, on which OS are you ?