SpinalHDL/VexRiscv #187: Illegal instruction exception

aignacio, 2021-06-11:
Hey @Dolu1990,

I'm using the cfg VexRiscvAxi4WithIntegratedJtag core and I'm seeing an illegal instruction exception when cache starts fetching first batch of instructions...I'm using the hello world program available also in VexRiscV repository. Can you take a look in the waves below and give me your opinion on that?
[image]
See the fst attached below. Also here's the assembly snippet that seems to be wrong:
[image]

[hello_world.zip](https://github.com/SpinalHDL/VexRiscv/files/6640225/hello_world.zip)
[waves.fst.zip](https://github.com/SpinalHDL/VexRiscv/files/6640223/waves.fst.zip)

Dolu1990, 2021-06-14:
Hi,

The issue seems to me to be in the memory interconnect or in the ram you read.

reason is that what the CPU read from its ibus do not match the assembly code : 
[image]

aignacio, 2021-06-16:
Hi @Dolu1990,
tks again, I checked and my elf loader was broken, fixed and now the CPU is working fine. Last question, is that possible to generate the smallest CPU version with AXI4 I/F?

Dolu1990, 2021-06-18:
Hi,

> the smallest CPU version with AXI4 I/F

Yes it should be. there is a to toAxi4ReadOnly for the iBus and a  toAxi4Shared for the dbus. Then on that toAxi4Shared, you can do a toAxi4 to have a regular AXI

aignacio, 2021-06-20:
tks!!

closed 2021-06-20 by aignacio