SpinalHDL/VexRiscv #188: SBT Debian packages no longer available

geertu, 2021-07-08:
The SBT repository referenced in README.md no longer works.

When running "apt update" (on Ubuntu 20.04LTS):
```
E: Failed to fetch https://dl.bintray.com/sbt/debian/InRelease  403  Forbidden [IP: 3.124.176.138 443]
E: The repository 'https://dl.bintray.com/sbt/debian  InRelease' is no longer signed.
```

Perhaps they have moved?

Dolu1990, 2021-07-08:
Hi,

I have no idea ^^

Is the following working for you ? 

```shell
echo "deb https://repo.scala-sbt.org/scalasbt/debian all main" | sudo tee /etc/apt/sources.list.d/sbt.list
echo "deb https://repo.scala-sbt.org/scalasbt/debian /" | sudo tee /etc/apt/sources.list.d/sbt_old.list
curl -sL "https://keyserver.ubuntu.com/pks/lookup?op=get&search=0x2EE0EA64E40A89B84B2DF73499E82A75642AC823" | sudo apt-key add
sudo apt-get update
sudo apt-get install sbt
```

geertu, 2021-07-08:
@Dolu1990 Yes it does, thanks!

tomverbeure, 2021-07-08:
Hey @Dolu1990, I just ran into the same issue while installing a new PC. Probably not a bad idea to update the front-end README?

@geertu, what a timely issue (and an unexpected one from a SW techie)! It's been, what, 25 years? :-) 

Tom

Dolu1990, 2021-07-09:
Updated ^^

closed 2021-07-09 by Dolu1990