SpinalHDL/VexRiscv #192: MMU on vexriscv - virtual to physical

Aljeshka, 2021-07-19:
Hey,

I build myself a vexriscv core on my arty and im trying to figure out how the DRAM addressing exactly works. When i use "sdram_force_wrphase 0x4FFFFFFF" i would expect that the DRAM adress would be something like ROW_BA_COL <= "FFFFFF". 
But thats definetly not the case. So my only guess is that the vexrisc uses virtual addresses with an MMU unit. Is that correct? If so, how do i convert the virtual address to physical?

Greetings
Alec

tomverbeure, 2021-07-19:
The VexRiscv only uses an MMU if you've configured it to use an MMU. Otherwise, it's physical addresses all the way up and down. You'll find a lot of different demo configurations in the [demo](https://github.com/SpinalHDL/VexRiscv/tree/master/src/main/scala/vexriscv/demo) directory. Most of them don't have an MMU, but some of them do. For example, the [GenFull demo has an MMU](https://github.com/SpinalHDL/VexRiscv/blob/a4c86130cc4fec7d86052ab9deab90b29c307a1d/src/main/scala/vexriscv/demo/GenFull.scala#L49-L52).

Aljeshka, 2021-07-19:
thanks for the quick answer and for the links. I only used the demo boards from the litex github. May i ask how to create a bitstream for one of the demos you send me?

Aljeshka, 2021-07-19:
nvm i guess like this "sbt "runMain vexriscv.demo.GenFull""

Aljeshka, 2021-07-19:
unfortunately none of these demos exploit dram memory. So my question still stands. Is there a way to translate the virtual address to physical for the MMU Plugin?

tomverbeure, 2021-07-19:
I've personally never use the VexRiscv MMU. Barebones CPUs are much easier to set up and work with. I don't know what your goals are, but if you have the option, I'd use a CPU that doesn't use an MMU.

Aljeshka, 2021-07-19:
My goal is simply to verify, that the physical address at the DRAM port is correct. For that i need to know what the physical address is :D

Aljeshka, 2021-07-19:
If you have some simple CPU example which exploits the dram without using MMU you, I would be very greatful

tomverbeure, 2021-07-19:
I do not: I have never used Litex. Maybe @Dolu1990 has an answer here.

Aljeshka, 2021-07-19:
I can also work with the mmu unit, but for that i need to know how it translates the virtual address to physical

Dolu1990, 2021-07-19:
Unless you run linux, which would initialise the MMU, there is no address translation. One thing about the wishbone based vexriscv, the address is in word, not in byte, maybe it is part of the trouble ?

What is sdram_force_wrphase 0x4FFFFFFF ?

Dolu1990, 2021-07-19:
(about the mmu, it won't start unless the software had initialise the STAP CSR, and switch to the supervisor mode)

Also, as far as i remember,  vexriscv.demo.GenFull do not implement the MMU, so a byte address on the register file will be translated as    address >> 2   on the wishbone bus. (normaly)