SpinalHDL/VexRiscv #194: Briey SoC is not debuggable over JTAG

nvitya, 2021-08-28:
Hi,

I've tested the Murax and the Briey SoCs.
I can upload and debug SW on the Murax (I've just increased the on chip RAM size to 32k).
The debugging does not work on the Briey. Sometimes after load you can step once or twice but then it does not stop.
The sw uploaded over JTAG definitely does not run.
I've tested with Cyclone IV E and Xilinx Atrix-7 too. I've even removed the VGA and SDRAM, the JTAG debugging still did not work.

It would be great if this worked. It is not so easy to create a proper multi-master bus crossbar. And the SDRAM definitely needs a cache system for the instruction fetching.

br.
nvitya

Dolu1990, 2021-08-31:
Hi,

One thing is, did you use the same cpu0.yaml in both test ? if yes, it is one issue.
(The one to use with the cached version need to have the instructions about how to flush the instruction cache)

Else, i'm not aware of any issue so far.

at which speed are you running briey and the JTAG ?

nvitya, 2021-09-01:
Hi,

The cpu0.yaml was always the same (the cached one):
```
debug: !!vexriscv.DebugReport {hardwareBreakpointCount: 0}
iBus: !!vexriscv.BusReport
  flushInstructions: [4111, 19, 19, 19]
  info: !!vexriscv.CacheReport {bytePerLine: 32, size: 4096}
  kind: cached
```
This worked with the Murax too.

Briey was running at 50 MHz.
The JTAG at 800 kHz.

Dolu1990, 2021-09-02:
Seems all good.

What board are you using for Atrix-7 ?
With some luck i have the same XD

nvitya, 2021-09-02:
I have this set:
https://de.aliexpress.com/item/32964497318.html?spm=a2g0o.store_pc_groupList.8148356.5.2b817fb297o38T

The resources including schematic can be donwloaded from here:
http://www.chinaqmtech.com/download_fpga
http://www.chinaqmtech.com/filedownload/38776

But I think any board is sufficient for this test.

Dolu1990, 2021-09-02:
I successfuly recreated some issues in simulation, i'm on it ^^

Dolu1990, 2021-09-02:
Issue fixed, that was the bridge from the dbus of the CPU to AXI which was picking a signal one stage above.
I pushed the fix to master  (https://github.com/SpinalHDL/VexRiscv/commit/cc9f3e753a2e1b7554e10bf43635f0d2b8afed82#diff-002b01c6e01d9aaa5bed395b343d572c97497b9f58c07f96e64f2d9cd6d4f5d9R292)

Got 

```shell
# Terminal 1
git clone https://github.com/SpinalHDL/VexRiscv.git
cd VexRiscv
sbt "runMain vexriscv.demo.Briey"
cd src/test/cpp/briey
make clean run

# Terminal 2
git clone https://github.com/SpinalHDL/openocd_riscv.git
cd openocd_riscv
./bootstrap
./configure --enable-ftdi --enable-dummy
make -j$(nproc)
src/openocd -f tcl/interface/jtag_tcp.cfg -c "set BRIEY_CPU0_YAML ../VexRiscv/cpu0.yaml" -f tcl/target/briey.cfg

# test 1 (was already OK)
telnet localhost 4444
> mdw 0x80000000 8
0x80000000: ???????? ????????????????????????
> mww 0x80000000 0x12345678
> mdw 0x80000000 8         
0x80000000: 12345678 ????????????????????????

# test 2 (that one was KO)
git clone https://github.com/SpinalHDL/VexRiscvSocSoftware.git
cd VexRiscvSocSoftware/projects/briey/
cd uart
make
riscv-none-embed-gdb build/uart.elf
target remote localhost:3333
set remotetimeout 60
set arch riscv:rv32
monitor reset halt
load
continue
```

Dolu1990, 2021-09-02:
Thanks for the issue report  :D

closed 2021-09-02 by Dolu1990