SpinalHDL/VexRiscv #195: Briey SDRAM Reverse Writing / Reading Error

nvitya, 2021-09-03:
Hi,

Now I can debug the Briey SoC over JTAG!  Tank you!

 Clock Speed: 50 MHz
 Test board: QMTECH Cyclone IV Starter Kit (SDRAM = W9825G6JH6)

Unfortunatelly the SDRAM does not work properly like with QSYS, when tested backwards.
It seems that the byte masking does not work somehow.

Here are the results of a 8-bit backward test with the Briey (2th Sep.):

```
SDRAM-8-reverse test 0 ...
  Filling...   clocks: 5157
  Comparing... 
ERR at 0x400003FF = 0x47 instead of 0x44 !
ERR at 0x400003FE = 0x47 instead of 0x45 !
ERR at 0x400003FD = 0x47 instead of 0x46 !
ERR at 0x400003FB = 0x4B instead of 0x48 !
ERR at 0x400003FA = 0x4B instead of 0x49 !
ERR at 0x400003F9 = 0x4B instead of 0x4A !
ERR at 0x400003F7 = 0x4F instead of 0x4C !
ERR at 0x400003F6 = 0x4F instead of 0x4D !
ERR at 0x400003F5 = 0x4F instead of 0x4E !
ERR at 0x400003F3 = 0x53 instead of 0x50 !
ERR at 0x400003F2 = 0x53 instead of 0x51 !
ERR at 0x400003F1 = 0x53 instead of 0x52 !
ERR at 0x400003EF = 0x57 instead of 0x54 !
ERR at 0x400003EE = 0x57 instead of 0x55 !
ERR at 0x400003ED = 0x57 instead of 0x56 !
ERR at 0x400003EB = 0x5B instead of 0x58 !
ERR at 0x400003EA = 0x5B instead of 0x59 !
ERR at 0x400003E9 = 0x5B instead of 0x5A !
ERR at 0x400003E7 = 0x5F instead of 0x5C !
ERR at 0x400003E6 = 0x5F instead of 0x5D !
ERR at 0x400003E5 = 0x5F instead of 0x5E !
ERR at 0x400003E3 = 0x63 instead of 0x60 !
ERR at 0x400003E2 = 0x63 instead of 0x61 !
ERR at 0x400003E1 = 0x63 instead of 0x62 !
ERR at 0x400003DF = 0x67 instead of 0x64 !
ERR at 0x400003DE = 0x67 instead of 0x65 !
ERR at 0x400003DD = 0x67 instead of 0x66 !
ERR at 0x400003DB = 0x6B instead of 0x68 !
ERR at 0x400003DA = 0x6B instead of 0x69 !
ERR at 0x400003D9 = 0x6B instead of 0x6A !
ERR at 0x400003D7 = 0x6F instead of 0x6C !
ERR at 0x400003D6 = 0x6F instead of 0x6D !
ERR at 0x400003D5 = 0x6F instead of 0x6E !
ERR at 0x400003D3 = 0x73 instead of 0x70 !
ERR at 0x400003D2 = 0x73 instead of 0x71 !
ERR at 0x400003D1 = 0x73 instead of 0x72 !
ERR at 0x400003CF = 0x77 instead of 0x74 !
ERR at 0x400003CE = 0x77 instead of 0x75 !
ERR at 0x400003CD = 0x77 instead of 0x76 !
ERR at 0x400003CB = 0x7B instead of 0x78 !
ERR at 0x400003CA = 0x7B instead of 0x79 !
ERR at 0x400003C9 = 0x7B instead of 0x7A !
ERR at 0x400003C7 = 0x7F instead of 0x7C !
ERR at 0x400003C6 = 0x7F instead of 0x7D !
ERR at 0x400003C5 = 0x7F instead of 0x7E !
ERR at 0x400003C3 = 0x83 instead of 0x80 !
ERR at 0x400003C2 = 0x83 instead of 0x81 !
ERR at 0x400003C1 = 0x83 instead of 0x82 !
ERR at 0x400003BF = 0x87 instead of 0x84 !
ERR at 0x400003BE = 0x87 instead of 0x85 !
ERR at 0x400003BD = 0x87 instead of 0x86 !
ERR at 0x400003BB = 0x8B instead of 0x88 !
ERR at 0x400003BA = 0x8B instead of 0x89 !
ERR at 0x400003B9 = 0x8B instead of 0x8A !
[... 262 lines cut ...]
ERR at 0x100FFF58 = 0xD3 instead of 0xEB !
ERR at 0x100FFF57 = 0xD2 instead of 0xEC !
ERR at 0x100FFF56 = 0xD3 instead of 0xED !
ERR at 0x100FFF55 = 0xD2 instead of 0xEE !
ERR at 0x100FFF54 = 0xD3 instead of 0xEF !
ERR at 0x100FFF53 = 0xD2 instead of 0xF0 !
ERR at 0x100FFF52 = 0xD3 instead of 0xF1 !
ERR at 0x100FFF51 = 0xD2 instead of 0xF2 !
ERR at 0x100FFF50 = 0xD3 instead of 0xF3 !
ERR at 0x100FFF4F = 0xD2 instead of 0xF4 !
ERR at 0x100FFF4E = 0xD3 instead of 0xF5 !
ERR at 0x100FFF4D = 0xD2 instead of 0xF6 !
ERR at 0x100FFF4C = 0xD3 instead of 0xF7 !
ERR at 0x100FFF4B = 0xD2 instead of 0xF8 !
ERR at 0x100FFF4A = 0xD3 instead of 0xF9 !
ERR at 0x100FFF49 = 0xD2 instead of 0xFA !
ERR at 0x100FFF48 = 0xD3 instead of 0xFB !
ERR at 0x100FFF47 = 0xD2 instead of 0xFC !
clocks: 12423847
  !!! ERRORS = 65
END SDRAM Test
```
The 8-bit and 16-bit forward tests work error-free on both Briey and QSYS.
The 32-bit backward test works error-free on both Briey and QSYS.
The 16-bit backward test works error-free on on QSYS with DBusSimplePlugin.

nvitya, 2021-09-03:
Sorry my fault !
The DQM signal was commented out.
Now the reverse testing also works with the Briey!.
Thank you for your support.

closed 2021-09-03 by nvitya

Dolu1990, 2021-09-03:
Cool  ^^
Welcome :)