SpinalHDL/VexRiscv #197: Debug RV32IMC

aignacio, 2021-09-13:
Hey all,
is that possible to debug a RV32IMC VexRISC-V w/JTAG? I'm currently having an issue that when the core hits a non-aligned jumped it halts and connection is lost with the debug adapter...the same issue is not present when the program is compiled with march=rv32i...

Dolu1990, 2021-09-14:
Hi,

Yes it should normaly have worked. But at the same time you are using a configuration that i haven't tried before ( no cache + RVC + AXI), so there may be something there.

A few questions : 
- Which git hash of VexRiscv / openocd_riscv are you using ?
- Is the CPU working with RVC (running a programm, not debugging it)
- Which SoC framework are you using ?

aignacio, 2021-09-14:
Hey @Dolu1990 

thanks for reply, here're the answers:

**Which git hash of VexRiscv / openocd_riscv are you using ?**

**VexRiscv**
> d67fe72de9f5ce288f2667b150314c926f4bbf4d 

**Riscv openocd - VexRISCV**
> c974c1b70348b59146bb87a1cfb829296240a509

**Is the CPU working with RVC (running a programm, not debugging it)**

> Yes, I tested the same program through Verilator model and it works fine

**Which SoC framework are you using ?**

> It's my own SoC, basically with the following cfg:
- Vex RISC-V core
- [AXI Slave] IRAM - 128KB
- [AXI Slave] DRAM - 128KB
- [AXI Slave] SPI Master
- AXI Interconnect 

I'm attaching the debug files I'm using to launch the debug adapter..

[Archive.zip](https://github.com/SpinalHDL/VexRiscv/files/7164901/Archive.zip)

Dolu1990, 2021-09-15:
So, i tried compressed instruction debug on Briey without caches and it worked well as far as i tried. (on the dev branch)

So, i can see that the VexRiscv version that you are using is a bit old, and do not has a few fixes (ex : https://github.com/SpinalHDL/VexRiscv/commit/cc9f3e753a2e1b7554e10bf43635f0d2b8afed82)

Can you give a try with the last master version ? or if you can, with the dev branch ? (which require SpinalHDL dev branch cloned in the parent directory)

Dolu1990, 2021-09-15:
let's me know how things goes :)

aignacio, 2021-09-16:
Hey @Dolu1990,

I tried with the master branch update and still the same =/....here's the output of my openocd command with the GDB on the right side, I'm launching gdb with the connection+loading:
```bash
$(GDB) $(TARGET).elf -ex "target extended-remote localhost:3333" -ex "load"
```

[image]

It seems weird that after the load (and even if I load again the elf), the memory content doesn't change. I'm running the design on a Xilinx FPGA so I tried to also add the ILA probing the memory AXI interface signals and the trigger never hit because there are no transactions coming from the only AXI master in the interconnect (VexRISCV core) during gdb `load` command.

[image]

Dolu1990, 2021-09-17:
So, one thing that you can do to test things with a smaller eco system (no gdb complexity) is to use the openocd telnet directly

Then you can test basic operations with less overhead. Ex : 
Read with mdw, write with mww.

mdw 0x80000000 8
mww 0x80000000 0x1234578
mdw 0x80000000 8

The issue is weird because the RVC nature of the programm should not interfer  if you just load the binary and check the memory content. So to be sure, can you confirm that if you keep the same RVC ready harwdare, but only change the compilation RVC flag it enable / disable the bug ?

Adding RVC capabilities to the hardware do make some decode path much longer. Are the FPGA timing passing the slack test ?

aignacio, 2021-09-17:
Timing is fine =), I'll another program just to see how it goes.
[image]

aignacio, 2021-09-17:
Hey @Dolu1990,

I tried again with/out the compressed instruction set and indeed I had the same behaviour, so I guess the first comment was a matter of luck =). So  I decided to take a different approach this time, building another vex core based on the demo example of VexRiscvAxi4wintegratedJtag that it's available in the master branch. This is the configuration I'm using now:
```scala
package vexriscv.demo

import spinal.core._
import spinal.lib._
import spinal.lib.bus.amba4.axi.Axi4ReadOnly
import spinal.lib.bus.avalon.AvalonMM
import spinal.lib.com.jtag.Jtag
import spinal.lib.eda.altera.{InterruptReceiverTag, QSysify, ResetEmitterTag}
import vexriscv.ip.{DataCacheConfig, InstructionCacheConfig}
import vexriscv.plugin._
import vexriscv.{VexRiscv, VexRiscvConfig, plugin}

/**
 * Created by spinalvm on 14.07.17.
 */
//class VexRiscvAvalon(debugClockDomain : ClockDomain) extends Component{
//
//}

object VexRiscvAxi4WithIntegratedJtag{
  def main(args: Array[String]) {
    val report = SpinalVerilog{

      //CPU configuration
      val cpuConfig = VexRiscvConfig(
        plugins = List(
          new PcManagerSimplePlugin(0x00000000l, false),
//          new IBusSimplePlugin(
//            interfaceKeepData = false,
//            catchAccessFault = false
//          ),
//          new DBusSimplePlugin(
//            catchAddressMisaligned = false,
//            catchAccessFault = false
//          ),
          new IBusCachedPlugin(
            prediction = STATIC,
            config = InstructionCacheConfig(
              cacheSize = 4096,
              bytePerLine =32,
              wayCount = 1,
              addressWidth = 32,
              cpuDataWidth = 32,
              memDataWidth = 32,
              catchIllegalAccess = true,
              catchAccessFault = true,
              asyncTagMemory = false,
              twoCycleRam = true,
              twoCycleCache = true
            ),
            compressedGen = true
            //            askMemoryTranslation = true,
            //            memoryTranslatorPortConfig = MemoryTranslatorPortConfig(
            //              portTlbSize = 4
            //            )
          ),
           new DBusSimplePlugin(
            catchAddressMisaligned = false,
            catchAccessFault = false
          ),
          //new DBusCachedPlugin(
            //config = new DataCacheConfig(
              //cacheSize         = 4096,
              //bytePerLine       = 32,
              //wayCount          = 1,
              //addressWidth      = 32,
[... 126 lines cut ...]
}
```
The changes I made compared to the file that comes in `src/main/scala/vexriscv/demo/VexRiscvAxi4WithIntegratedJtag.scala` are:
1) Removed the cache on the dbus I/F (it was causing the core to dispatch axi bursts when writing to my peripherals, what's not supported);
2) Changed regFile plugin to use ASYNC instead of SYNC once based on a previous ticket you mention it's not compatible;
3) Added compressedGen = true to the configuration;

The issue that I'm having still persists, I can start debugging but sometimes it drops in the first step or sometimes when there's a jump in the first instructions...

[image]

Searching through the issues it seems pretty much like this guy comment:
https://github.com/SpinalHDL/VexRiscv/issues/194
```
...Sometimes after load you can step once or twice but then it does not stop.
```
At least now I've tried to write to the peripherals through GDB and it's working, I'm not sure what I'm missing here to make it work...do you have a configuration with the following features tested?

1) debug pluggin - jtag
2) AXI4
3) compressedGen = true

If so I can try to run another synthesis using this core and see how it goes. Another funny aspect of this issue is that through my simulations (which I don't use JTAG, I load by dpi methods the fw) all the cores I've tried so far are working perfectly....

Ps.: When using dbusCachedPlugin is there a way to force the cache to not dispatch a burst during reads/writes?

Dolu1990, 2021-09-17:
> Removed the cache on the dbus I/F (it was causing the core to dispatch axi bursts when writing to my peripherals, what's not supported); /// Ps.: When using dbusCachedPlugin is there a way to force the cache to not dispatch a burst during reads/writes?

You can adjust ioRange      = _(31 downto 28) === 0xF    to specify which address range should be accessed by the CPU in a non bursted  and uncached mode.   _ represent the address and then you can do what ever test on it which return true when is should be unbursted and uncached

> do you have a configuration with the following features tested?

Yes, the ones which run linux with fpu and all those things, but that's a bit heavy.
You can try the Briey config.  which is composed of the followed plugins : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/Briey.scala#L53
+ https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/Briey.scala#L290

You will have to turn one RVC in the IBusCachedPlugin and turn the registerfile to ASYNC. That's the config i tested the other days with debug.

If you want there is also some verilator code to connect openocd to the simulation :)

aignacio, 2021-09-17:
Thanks for quick reply, I'll for sure take this cfgs!, can you tell me more about this way of connecting openocd with verilator? All my sys are using verilator =)

Dolu1990, 2021-09-17:
so, for the openocd -> verilator : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/test/cpp/briey/main.cpp#L379
https://github.com/SpinalHDL/VexRiscv/blob/master/src/test/cpp/common/jtag.h

Basicaly, this is based on a little abstraction layer above verilator which allow the tick function to be rescheduled in a future time :
https://github.com/SpinalHDL/VexRiscv/blob/master/src/test/cpp/common/jtag.h#L174

Else, there is the equivalent in SpinalSim (may help to understand) : 
https://github.com/SpinalHDL/SpinalHDL/blob/b42ad071474c2042bf13d9ff30147b0e378b322d/lib/src/main/scala/spinal/lib/com/jtag/sim/JtagTcp.scala#L11

Dolu1990, 2021-09-17:
Then instead of using your jtag adapter script for openocd, you can use  tcl/interface/jtag_tcp.cfg

aignacio, 2021-09-18:
Thanks @Dolu1990,

I finally found the issue, the problem was related to the SPI Master peripheral once by default the clk generation was wrong causing the CTS (clock tree synthesis) of the FPGA to be unbalanced and holding it low for some clk cycles, putting then the vex core into an unknown state, to find the issue I tweak the peripheral before launching the program to run and it worked. Eventually it was impossible to see this scenario through verilator because there's no fanout in the clk there =)

closed 2021-09-18 by aignacio

Dolu1990, 2021-09-18:
Hoo cool :)

gg ^^

Saqlinahamad, 2021-12-03:
Hi @Dolu1990 @aignacio
My problem is also same and had raised a new issue in repo 
Please just check and help accordingly to the issue🙂