SpinalHDL/VexRiscv #201: could I use Murax of VexRiscv + picorv32 firmware

fatalfeel, 2021-09-27:
iceprog toplevel.bin in Murax of VexRiscv (gateware)
iceprog firmware.bin in picorv32
can do them both?
burn the gateware then firmware
then I can add instruction in Murax and modify riscv gcc compiler
then modify the firmware to test new instruction

Dolu1990, 2021-09-27:
Hi,

I'm not sure of the whole context.  firmware.bin is some software made for picorv32 ?

fatalfeel, 2021-09-27:
yes picorv32
https://github.com/cliffordwolf/picorv32.git

my build step
http://fatalfeel.blogspot.com/2021/09/risc-v-on-icesugar.html

Dolu1990, 2021-09-27:
Much of it should be fine, but if you use interrupts, it won't work, because picorv32 and vexriscv have different interrupt implmentations.

Also, you have to be sure that the reset vector is the same, or you have to compile the binary to be position independant

fatalfeel, 2021-09-27:
thanks u first
just using simple way test no irq. ca do it well?
https://github.com/wuxx/icesugar/tree/master/src/advanced/picorv32/scripts/icestorm

Dolu1990, 2021-09-28:
normaly, if you program is placed in the onchipram, you only need to burn the bitstream, as it should contains the firmware.

See https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/Murax.scala#L476

fatalfeel, 2021-09-28:
thanks a lot, you mean hello world bind to ram and  burn at once right?

Dolu1990, 2021-09-28:
> thanks a lot, you mean hello world bind to ram and burn at once right?

Yes

> src/main/c/murax/hello_world/build/hello_world.hex

So you need to build your binary and generate the .hex from it. Then point the path to your hex when you generate the hardware.

This will generate the proper Murax.v_toplevel_system_ram_ram_symbol?.bin

when you want to debug /develop things, the best maybe to just use the jtag to load the binaries you want without regenerating and flashing everything.

fatalfeel, 2021-09-28:
> Yes
>
>
> So you need to build your binary and generate the .hex from it. Then point the path to your hex when you generate the hardware.
>
> This will generate the proper [...]

very very very thank you~~~~the best Idea you give~~~

another question :
in ~/VexRiscv/src/main/c/murax/hello_world/src/main.c and murax.h
there is a define GPIOA 0xF0000000
1. is GPIOA mapping to murax.io.gpioA ?
2. why define pointer 0xF0000000, is it a address?

Dolu1990, 2021-09-29:
> is GPIOA mapping to murax.io.gpioA ?

Yes

> why define pointer 0xF0000000, is it a address?

Yes it is, in Murax, all peripheral are in the 0xF??????? range

> how to know it in 0xF0020010

Only way, is to check the Murax.scala. You have to cummulate the apbBridge address with the peripheral address

fatalfeel, 2021-09-29:
> > is GPIOA mapping to murax.io.gpioA ?
> 
> Yes
> 
> > why define pointer 0xF0000000, is it a address?
> 
> Yes it is, in Murax, all peripheral are in the 0xF??????? range
> 
> > how to know it in 0xF0020010
> 
> Only way, is to check the Murax.scala. You have to cummulate the apbBridge address with the peripheral address

//You Are My Sunshine , thank u
https://www.youtube.com/watch?v=dh7LJDHFaqA

closed 2021-09-30 by fatalfeel

Dolu1990, 2021-09-30:
Cool ^^

nerdylye, 2025-04-18:
Hi @fatalfeel, I'm trying to do the same thing as you did. Do you mind sharing how did you get it to work? Thanks a lot! :D