SpinalHDL/VexRiscv #202: PMP: Address range with NAPOT

occheung, 2021-09-30:
# One-line summary
The `PMPPlugin` is unable to determine the address protection range correctly when using NAPOT mode.

# Step to reproduce
- Generate VexRiscv CPU with the following PMPPlugin, using the latest master.
  ```
  PmpPlugin(
      regions = 16,
      granularity = 256,
      ioRange = _.msb
  )
  ```
- Configure a PMP region of size 256 using NAPOT in machine mode, for example:
  - Base address: `0x4015bf00`, then set `pmpaddr0 = 0x10056fdf`
  - Write `pmpcfg0 = 0x98` (only setting NAPOT and the lock bit, and disable read/write/execute)
- Read memory from `0x4015bff0`
  - No load error, despite being part of the protected region 0.
- Repeat the above step with `0x4015bfe0`, `0x4015bfd0`, `0x4015bfc0`, ..., `0x4015bf80`. The result should be the same.
- Finally, a load fault is raised when `0x4015bf70` is read.

The value in `pmpaddr0` has 5 continuous ones at the end, so it should represent a protection region of 2^(5+3) = 256 right?

occheung, 2021-09-30:
Related: #174 
Also, the granularity of PMP access control is violated.

Dolu1990, 2021-09-30:
Hi,

So, the current implementation in dev was optimized a few months ago to use distributed ram to store the PMP states (reduce area) But in the process NAPOT was dropped.

For that reason, i added in the dev branch the Pmp implementation pre optimization, which support NAPOT : 
https://github.com/SpinalHDL/VexRiscv/blob/dev/src/main/scala/vexriscv/plugin/PmpPluginOld.scala

I'm not so much aware of the PMP spec, what do you mean by : 
"Also, the granularity of PMP access control is violated." ? 

Thanks anyway ^^

occheung, 2021-09-30:
Here is a paragraph in RISC-V privileged manual regarding the granularity.
[image]
A granularity of 256 should imply a smallest configurable NAPOT size is 256. The behavior shown up there seems to not match the specification.

I think the issue is that there are 1 bit too many for the address base, due to the 0 separator between the terminating "1"s and the base address. I tried the following patch to fix this behavior for my case.
```
diff --git a/src/main/scala/vexriscv/plugin/PmpPlugin.scala b/src/main/scala/vexriscv/plugin/PmpPlugin.scala
index 1084891..73c4c78 100644
--- a/src/main/scala/vexriscv/plugin/PmpPlugin.scala
+++ b/src/main/scala/vexriscv/plugin/PmpPlugin.scala
@@ -85,12 +85,18 @@ trait Pmp {
 class PmpSetter(cutoff : Int) extends Component with Pmp {
   val io = new Bundle {
     val addr = in UInt(xlen bits)
-    val base, mask = out UInt(xlen - cutoff bits)
+
+    // Can skip 1 more bit due to the guaranteed 0 separating address base and
+    // the lower order bit ones.
+    // Since granularity is at least 8, we can ignore the NA4 case, where the
+    // zero-separator is non-existent.
+    //
+    // Also, the base is exactly a mask
+    val base = out UInt(xlen - cutoff - 1 bits)
   }
 
   val ones = io.addr & ~(io.addr + 1)
-  io.base := io.addr(xlen - 3 downto cutoff - 2) ^ ones(xlen - 3 downto cutoff - 2)
-  io.mask := ~ones(xlen - 2 downto cutoff - 1)
+  io.base := (io.addr(xlen - 3 downto cutoff - 2) ^ ones(xlen - 3 downto cutoff - 2))(xlen - cutoff - 1 downto 1)
 }
 
 case class ProtectedMemoryTranslatorPort(bus : MemoryTranslatorBus)
@@ -127,7 +133,7 @@ class PmpPlugin(regions : Int, granularity : Int, ioRange : UInt => Bool) extend
     val state = pipeline plug new Area {
       val pmpaddr = Mem(UInt(xlen bits), regions)
       val pmpcfg = Vector.fill(regions)(Reg(Bits(8 bits)) init (0))
-      val base, mask = Vector.fill(regions)(Reg(UInt(xlen - cutoff bits)))
+      val base = Vector.fill(regions)(Reg(UInt(xlen - cutoff - 1 bits)))
     }
 
     def machineMode : Bool = privilegeService.isMachine()
@@ -245,7 +251,6 @@ class PmpPlugin(regions : Int, granularity : Int, ioRange : UInt => Bool) extend
         
         when (fsmEnable & ~state.pmpcfg(fsmCounter)(lBit)) {
           state.base(fsmCounter) := setter.io.base
-          state.mask(fsmCounter) := setter.io.mask
         }
       }
     }
@@ -253,7 +258,7 @@ class PmpPlugin(regions : Int, granularity : Int, ioRange : UInt => Bool) extend
     pipeline plug new Area {
       def getHits(address : UInt) = {
         (0 until regions).map(i =>
-            ((address & state.mask(U(i, log2Up(regions) bits))) === state.base(U(i, log2Up(regions) bits))) &
+            (address === state.base(U(i, log2Up(regions) bits))) &
             (state.pmpcfg(i)(lBit) | ~machineMode) & (state.pmpcfg(i)(aBits) === NAPOT)
         )
       }
@@ -272,7 +277,7 @@ class PmpPlugin(regions : Int, granularity : Int, ioRange : UInt => Bool) extend
         dPort.bus.rsp.allowExecute := False
         dPort.bus.busy := False
 
-        val hits = getHits(address(31 downto cutoff))
+        val hits = getHits(address(31 downto cutoff + 1))
 
         when(~hits.orR) {
           dPort.bus.rsp.allowRead := machineMode
@@ -294,7 +299,7 @@ class PmpPlugin(regions : Int, granularity : Int, ioRange : UInt => Bool) extend
         iPort.bus.rsp.allowWrite := False
         iPort.bus.busy := False
 
-        val hits = getHits(address(31 downto cutoff))
+        val hits = getHits(address(31 downto cutoff + 1))
 
         when(~hits.orR) {
           iPort.bus.rsp.allowExecute := machineMode

```
I think it worked in my case. However, I am not sure if it is the right thing to do.

Anyway, is the old implementation (`PmpPluginOld`) going to stay? Or is it just here temporarily?

Dolu1990, 2021-09-30:
> I think it worked in my case. However, I am not sure if it is the right thing to do.

@lindemer 

> Anyway, is the old implementation (PmpPluginOld) going to stay? Or is it just here temporarily?

As long as there is a reason to keep it, yes, also, it would fit well in ASIC, so i guess it will likely stay there for ever ever <3

occheung, 2021-10-01:
> I think it worked in my case. However, I am not sure if it is the right thing to do.

@lindemer
Quite certain it is **not** the right thing to do, given that I wasn't using a proper address mask for deciding hit/miss.
My patch will very likely apply PMP restrictions to unspecified memory sections erroneously.

@Dolu1990 
Thanks for the response. I think I will just use the old plugin then, since the granularity feature is unnecessary for my applications.