SpinalHDL/VexRiscv #204: what arch of riscv gcc for arty A7

fatalfeel, 2021-10-01:
in hello_world/makefile
RISCV_NAME ?= riscv64-unknown-elf
RISCV_PATH ?= /opt/riscv/

//build my gcc tool chain
git clone https://github.com/riscv/riscv-gnu-toolchain 
cd $HOME/icesugar/riscv-gnu-toolchain
git checkout 411d134
git submodule update --init --recursive
mkdir build
cd build
../configure --prefix=/opt/riscv64i --with-arch=rv64im

is rv64im fit for Murax/arty_a7 ?  or other arch parameter?

Dolu1990, 2021-10-01:
Hoo i used : 

```
version=riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-ubuntu14
wget -O riscv64-unknown-elf-gcc.tar.gz riscv https://static.dev.sifive.com/dev-tools/$version.tar.gz
tar -xzvf riscv64-unknown-elf-gcc.tar.gz
sudo mv $version /opt/riscv
echo 'export PATH=/opt/riscv/bin:$PATH' >> ~/.bashrc

```

Else rv32im should be good, ilp32 is good, you will just have to fix RISCV_NAME ?= riscv64-unknown-elf

to what it gives you (probably riscv32-unknown-elf)

fatalfeel, 2021-10-02:
thanks man

closed 2021-10-02 by fatalfeel