SpinalHDL/VexRiscv #209: could I use ARM-USB-TINY-H  Jtag debug murax

fatalfeel, 2021-10-09:
# write_debug_probes -force "~/riscv_cpu/VexRiscv/debug_probe.ltx"
INFO: [Chipscope 16-244] No debug cores were found in this design.
/root/~/riscv_cpu/VexRiscv/debug_probe.ltx

I want use ARM-USB-TINY-H Jtag connect JD Pin watch cpu running
can I do it?

Dolu1990, 2021-10-11:
Normaly, you would "just" need to bring the jtag wires of murax to that pmod, and put them to the arm usb tiny h dongle
Don't forget to put the dongle vref to 3.3V, and use the corresponding openocd jtag script.

if you want to use the BSCAN2, right, there will be no more jtag pins on the toplevel, and you will also need to use the corresponding openocd jtag script.

See for example : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/bsp/digilent/ArtyA7SmpLinux/openocd/usb_connect.cfg#L19

(ArtyA7 35-t)

Dolu1990, 2021-10-11:
Seems good to me.

Also, found some documentation about all of that : 
https://github.com/SpinalHDL/VexRiscv/tree/master/doc/nativeJtag

fatalfeel, 2021-10-12:
done and can set breakpoint full debug, thanks a lot, a lot

internal jtag eclipse setting:
https://www.mediafire.com/view/cwqifkpv4d27rgk/artya7-35t_internal_eclipse00.png
https://www.mediafire.com/view/jbd9r7f7rcc3y66/artya7-35t_internal_eclipse01.png
https://www.mediafire.com/view/fo5tlqccr8kijmg/artya7-35t_internal_eclipse02.png
https://www.mediafire.com/view/l81x5g1i4osxr3h/artya7-35t_internal_eclipse03.png

demo:
https://www.mediafire.com/view/tm67r0tv1vwe4il/artya7-35t_internal_demo00.png
https://www.mediafire.com/view/iua7omk5bittf69/artya7-35t_internal_demo01.png

full step in chpater [arty a7 internal usb jtag debug]
http://fatalfeel.blogspot.com/2013/12/risc-v-on-arty-a7-35t.html

fatalfeel, 2021-10-12:
//share my toplevel.v

`timescale 1ns / 1ps

module toplevel(
    input  wire clk100,
    input  wire cpu_reset,//active low

//external jtag start	
    /*input  wire tck,
    input  wire tms,
    input  wire tdi,
    input  wire trst,//ignored
    output reg  tdo,*/
//external jtag end

    input  wire serial_rx,
    output wire serial_tx,

    input  wire user_sw0,
    input  wire user_sw1,
    input  wire user_sw2,
    input  wire user_sw3,

    input  wire user_btn0,
    input  wire user_btn1,
    input  wire user_btn2,
    input  wire user_btn3,

    output wire user_led0,
    output wire user_led1,
    output wire user_led2,
    output wire user_led3
  );

  wire [31:0] io_gpioA_read;
  wire [31:0] io_gpioA_write;
  wire [31:0] io_gpioA_writeEnable;

  wire io_asyncReset = ~cpu_reset;

  assign {user_led3,user_led2,user_led1,user_led0} = io_gpioA_write[3 : 0];
  assign io_gpioA_read[3:0]     = {user_sw3,user_sw2,user_sw1,user_sw0};
  assign io_gpioA_read[7:4]     = {user_btn3,user_btn2,user_btn1,user_btn0};
  //assign io_gpioA_read[11:8]  = {tck,tms,tdi,trst}; //external jtag

//external jtag start	
  /*reg  tesic_tck,tesic_tms,tesic_tdi;
  wire tesic_tdo;
  reg  soc_tck,soc_tms,soc_tdi;
  wire soc_tdo;

  always @(*) begin
      {soc_tck,  soc_tms,  soc_tdi  } = {tck,tms,tdi};
      tdo = soc_tdo;
  end*/
//external jtag end

  Murax core (
    .io_asyncReset(io_asyncReset),
    .io_mainClk (clk100 ),
//external jtag start
	/*.io_jtag_tck(soc_tck),
    .io_jtag_tdi(soc_tdi),
    .io_jtag_tdo(soc_tdo),
    .io_jtag_tms(soc_tms),*/
//external jtag end
    .io_gpioA_read       (io_gpioA_read),
    .io_gpioA_write      (io_gpioA_write),
    .io_gpioA_writeEnable(io_gpioA_writeEnable),
    .io_uart_txd(serial_tx),
    .io_uart_rxd(serial_rx)
  );
endmodule