SpinalHDL/VexRiscv #216: Briey SoC simulation hang at beginning

Flywing2018, 2021-11-02:
Hi,
I follow the README to config/build/run verilator simulation for Briey SoC. The VGA window prompted.
However, it displays nothing, terminal print 2 lines below then no output any more.
[image]

Flywing2018, 2021-11-02:
BTW, I have try this flow both on Ubuntu20.04 and RHEL8, the result are same.
Does the Briey SoC flow miss something in README or I miss some important step?

Dolu1990, 2021-11-03:
Hi, i think it is normal, you need to load some binaries into the simulation, you can do it by connecting openocd and gdb to it
(the sdram is blank when the sim boot, and there is no bootloader in the onchip ram)