SpinalHDL/VexRiscv #217: Problem with package pkg_scala2hdl when simulating created processor in VHDL

BLangOS, 2021-11-03:
Hi,

from the vexrisc description `VexRiscvWB_small` shown below (small cpu, wishbone bus, external interrupts) a vhdl-model can be generated without errors. But when simulation the model in Modelsim I got the following error message: 
```
#** Fatal: (vsim-3734) Index value 0 is out of range 1 to 2.
#    Time: 0 ps  Iteration: 0  Process: /vexriscvwb_top_tb/DUT/MCU/CPU/line__2183 File: VexRiscv.vhd
# Fatal error in Subprogram pkg_extract at VexRiscv.vhd line 193
```

The error message is due to the fist function `pkg_extract` in package `body pkg_scala2hdl`, which is part of the generated vhdl file:
```
  function pkg_extract (that : std_logic_vector; bitId : integer) return std_logic is
  begin
    return that(bitId);
  end pkg_extract;
```
This function does not renumber the elements of the parameter vector `that`.
Thus I changed the function to the following code which eliminates the error:
```
  function pkg_extract (that : std_logic_vector; bitId : integer) return std_logic is
    variable ret : std_logic_vector(that'length-1 downto 0);
  begin
    ret := that;
    return ret(bitId);
  end pkg_extract;
```
As far as I see now my system works correctly. 
Is this the correct way to fix the error? If yes, maybe the package `pkg_scala2hdl` should be changed.

Regards, Bernhard. 

```
package vexriscv.demo

import vexriscv.plugin._
import vexriscv.{VexRiscv, VexRiscvConfig, plugin}
import spinal.core._
import spinal.lib._

object VexRiscvWB_small{
  def main(args: Array[String]) {
    //val report = SpinalConfig(mode = Verilog).generate{
    val report = SpinalConfig(mode = VHDL).generate{

      //CPU configuration
      val cpuConfig = VexRiscvConfig(
        plugins = List(
          new IBusSimplePlugin(
            resetVector = 0x00010000l,
            cmdForkOnSecondStage = false,
            cmdForkPersistence = false,
            prediction = NONE,
            catchAccessFault = false,
            compressedGen = true
          ),
          new DBusSimplePlugin(
            catchAddressMisaligned = false,
            catchAccessFault = false
          ),
          new CsrPlugin(
            CsrPluginConfig(
              catchIllegalAccess = false,
              mvendorid          = null,
              marchid            = null,
              mimpid             = null,
              mhartid            = null,
              misaExtensionsInit = 66,
              misaAccess         = CsrAccess.NONE,
              mtvecAccess        = CsrAccess.READ_WRITE, // required for xtvecModeGen
              mtvecInit          = 0x00010010l,
              mepcAccess         = CsrAccess.READ_WRITE, // required to access EPC from software 
              mscratchGen        = false,
              mcauseAccess       = CsrAccess.READ_ONLY,
              mbadaddrAccess     = CsrAccess.NONE,
              mcycleAccess       = CsrAccess.NONE,
              minstretAccess     = CsrAccess.NONE,
              ecallGen           = true,                   // enable
              wfiGenAsWait       = false,
              ucycleAccess       = CsrAccess.NONE,
              uinstretAccess     = CsrAccess.NONE,
              xtvecModeGen       = true
            )
          ),
          new DecoderSimplePlugin(
            catchIllegalInstruction = false
          ),
          new RegFilePlugin(
            regFileReadyKind = plugin.SYNC,
            zeroBoot = false
          ),
          new IntAluPlugin,
          new SrcPlugin(
            separatedAddSub = false,
            executeInsertion = false
          ),
          new LightShifterPlugin,
          new HazardSimplePlugin(
            bypassExecute           = false,
            bypassMemory            = false,
            bypassWriteBack         = false,
            bypassWriteBackBuffer   = false,
            pessimisticUseSrc       = false,
            pessimisticWriteRegFile = false,
            pessimisticAddressMatch = false
          ),
          new DebugPlugin(ClockDomain.current.clone(reset = Bool().setName("debugReset"))),
          new BranchPlugin(
            earlyBranch = false,
            catchAddressMisaligned = false
          ),
          new UserInterruptPlugin(
            interruptName = "LocalInt0",
            code = 16
          ),
          new UserInterruptPlugin(
            interruptName = "LocalInt1",
            code = 17
          ),
          new YamlPlugin("cpu0.yaml")
        )
      )

      //CPU instanciation
      val cpu = new VexRiscv(cpuConfig)

      //CPU modifications to be an AhbLite3 one
      cpu.rework {
        for (plugin <- cpuConfig.plugins) plugin match {
          case plugin: IBusSimplePlugin => {
            plugin.iBus.setAsDirectionLess() //Unset IO properties of iBus
            master(plugin.iBus.toWishbone()).setName("iBusWishbone")
          }
          case plugin: DBusSimplePlugin => {
            plugin.dBus.setAsDirectionLess()
            master(plugin.dBus.toWishbone()).setName("dBusWishbone")
          }
          case _ =>
        }
      }
      cpu
    }
  }
}
```

Dolu1990, 2021-11-03:
Hi,

Thanks :)

So, i guess you used VexRiscv master branch ? Because the VexRiscv dev branch use the SpinalHDL dev (1.6.1) which should have that issue fixed by using an alias.

To use VexRiscv dev branch, you also need to also clone the SpinalHDL dev branch in the parent directory of VexRiscv

Regards
Charles

BLangOS, 2021-11-04:
Hi Charles,
yes I did use the master branch up to now.
Due to your comment I loaded the most recent dev branches of VexRiscv and SpinalHDL and generated the vhdl-File of my `VexRiscvWB_small`-processor with this environnement. When simulating the generated VexRiscv.vhd the problem remained:
[image]
(FYI: I renamed the generated file `VexRiscv.vhd` to `VexRiscvWB_small.vhd`)
I am sure that now I use SpinalHDL dev (1.6.1), as you can see in the header of the generated file:
[image]
As I told, with my woraround everything works fine, but I must manually change the generated file.
If it helps you I can send you a complete modelsim project. 
Regards
Bernhard.

Dolu1990, 2021-11-04:
Hooo, damned, i was looking at the pkg_extract which use bit range, not single bit XD

It should now be fixed with the above commit in SpinalHDL dev :)

let's me know how it goes.

Thanks :D
Charles

BLangOS, 2021-11-04:
Yes, now it is fixed.
Thank you. :-)
Regards
Bernhard.

closed 2021-11-04 by BLangOS