SpinalHDL/VexRiscv #218: println function not working

nikita890-art, 2021-11-08:
Hi...
I am trying to execute this code,
.
.
.
void print(const char*str){
	while(*str){
		uart_write(UART,*str);
		str++;
	}
}
void println(const char*str){
	print(str);
	uart_write(UART,'\n');

  int main()
    {
    char message[1024]="what's your name\n";
        GPIO_A->OUTPUT_ENABLE = 0x0000000F;
	GPIO_A->OUTPUT = 0x00000001;
    println("what's your name");
.
.
.
 When I dump this code in the arty board, it gives some random output like a special character.
So, I try to run it on spike and I get a segmentation fault.

What can I do?

Note: This is just the snippet of the code. The actual code is kinda big. I dont think the entire code matters here but just in case if needed, I can upload it in the drive.

Dolu1990, 2021-11-08:
Hi,

A few question :
- Which soc are you using ?
- Which toplevel synthesis scripts are you using ?
- Did you checked the uart output on a scope or a simulation ? 

This won't work on spike i guess, as it does not implement the same peripherals.

nikita890-art, 2021-11-09:
1. the board I  am using is arty a7 35T
2. I am not sure about the synthesis scripts but we are performing synthesis on vivado
3. no
@Dolu1990

nikita890-art, 2021-11-09:
Is there any way I can get the message printed? @Dolu1990

Dolu1990, 2021-11-09:
I would say, the best is to scope the uart to understand the symptoms better.

Are you sure the frequency of the clock match the one specified at the SpinalHDL generation ?
The uart divider get a value depending on it.