SpinalHDL/VexRiscv #219: Readback of hardware breakpoint values via debug interface

BLangOS, 2021-11-09:
I am working on a hardware that uses the VexRiscv debug interface. Up to now it can halt, step and continue the processor, memory accesses work and it can set hardware breakpoints starting at address 0x40 of the debug interface.
Now I try to read back the hardware breakpoint values but it seems that this is not working.
Is it possible to configure the debug interface such that the breakpoint values (address and enable bit) can be read?
Regards Bernhard.

Dolu1990, 2021-11-10:
Hi,

Right, to save area, the hardware breapoints are write only : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/DebugPlugin.scala#L275

There is no setting to make them readable, but if you want you can try to implement it, it shouldn't be tricky.

Regards
Charles

BLangOS, 2021-11-10:
Hi Charles,

I did implement it in a way, that it is backward compatible.

In file `DebugPlugin.scala` I add a parameter `BreakpointReadback : Boolean = false` to `class DebugPlugin`.

The `when` statement starting from line 244 (see [https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/DebugPlugin.scala#L244](url)) I changed to 
```
      switch(RegNext(io.bus.cmd.address(7 downto 2))) {
        is(0x0) {
          io.bus.rsp.data(0) := resetIt
          io.bus.rsp.data(1) := haltIt
          io.bus.rsp.data(2) := isPipBusy
          io.bus.rsp.data(3) := haltedByBreak
          io.bus.rsp.data(4) := stepIt
        }
        if (BreakpointReadback) {
          for(i <- 0 until hardwareBreakpointCount){
            is(0x10 + i){
              io.bus.rsp.data(31 downto 1) := hardwareBreakpoints(i).pc.asBits
              io.bus.rsp.data(0)           := hardwareBreakpoints(i).valid
            }
          }
        }
      }

```
Now it is working. In case of readback the debug plugin must be called with `BreakpointReadback = true`, e.g.: 
```
          ...
          new DebugPlugin(
            ClockDomain.current.clone(reset = Bool().setName("debugReset")),
            hardwareBreakpointCount = 4,
            BreakpointReadback      = true
          ),
          ...
``` 
If the `BreakpointReadback` ist omitted or set to `false` I belive it is equivalent to the current code.

Is it possible that you add this change to the `dev`-Branch?
Here is the modified File: [DebugPlugin.zip](https://github.com/SpinalHDL/VexRiscv/files/7515959/DebugPlugin.zip)

If you are interested here are a few words about my project. I work on a GDB-Server in hardware. It is already running via serial port and can directly connect to gdb, but up to now only with software breakpoints. When trying to support hardware breakpoints I realized that the RSP-protocol of the gdb does not send breakpoint numbers but only breakpoint addresses. Thus I must check the current hardware breakpoints when they must be deleted or when new breakpoints must be set. Without readback I have to add a memory for the hardware breakpoints to my hardware what I want to avoid.
Soon I will publish the code of this server.

Regards Bernhard.

BLangOS, 2021-11-11:
I now proposed a pull request that is sligthly different to the above described change.
The proposed change in the pull request lets the DebugPlugin exactly as it is now when no readback is wished. In case of readback the additional hardware is added.

Regards Bernhard.

closed 2021-11-11 by BLangOS

Dolu1990, 2021-11-15:
Done ^^

Thanks :D