SpinalHDL/VexRiscv #221: linux image generation

sjthales, 2021-11-17:
Hello,

I am currently trying to generate linux image to simulate linux booting on VexRiscv.

To do so, I follows guidelines in src/main/scala/vexriscv/demo/Linux.scala:
```
Buildroot =>
git clone https://github.com/SpinalHDL/buildroot.git -b vexriscv
cd buildroot
make spinal_vexriscv_sim_defconfig
make 

```
But it doesn't work, I get the following error:

```
mkdir -p /home/sjacq/Work_dir/USE_CASE/2021/softcore_optimization/vexriscv/buildroot/output/target/etc
( \
	echo "NAME=Buildroot"; \
	echo "VERSION=2019.05-git-00630-g82656af66f"; \
	echo "ID=buildroot"; \
	echo "VERSION_ID=2019.05-git"; \
	echo "PRETTY_NAME=\"Buildroot 2019.05-git\"" \
) >  /home/sjacq/Work_dir/USE_CASE/2021/softcore_optimization/vexriscv/buildroot/output/target/usr/lib/os-release
ln -sf ../usr/lib/os-release /home/sjacq/Work_dir/USE_CASE/2021/softcore_optimization/vexriscv/buildroot/output/target/etc
>>>   Sanitizing RPATH in target tree
/home/sjacq/Work_dir/USE_CASE/2021/softcore_optimization/vexriscv/buildroot/support/scripts/fix-rpath target
>>>   Copying overlay board/spinal/vexriscv_sim/overlay
rsync: change_dir "/home/sjacq/Work_dir/USE_CASE/2021/softcore_optimization/vexriscv/buildroot//board/spinal/vexriscv_sim/overlay" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
make[1]: *** [Makefile:765 : target-finalize] Erreur 23
make: *** [Makefile:84 : _all] Erreur 2
```
What did I do wrong? What is the way to generate linux image?

Dolu1990, 2021-11-18:
It is because of the overlay that i added, but which has no content.
I just pushed a fix : 
https://github.com/SpinalHDL/buildroot/commit/30d350501a940ba1852d3f302cdaf8c0495c12f4

Should be good now :)

Dolu1990, 2021-11-18:
Keep in mind, those linux builds are kind of very old.

Currently the dev i'm on are based on :
https://github.com/SpinalHDL/SaxonSoc/tree/dev-0.3/bsp/digilent/ArtyA7SmpLinux
https://www.youtube.com/watch?v=K5Vv32O9g7Q&ab_channel=CharlesPapon

sjthales, 2021-11-18:
Thank you.

Linux image generation is now working.

However, now I encounter a problem in running linux simulation.

To do so, I ran the following commands:
```
Run linux in simulation (Require the machime mode emulator compiled in SIM mode) =>
sbt "runMain vexriscv.demo.LinuxGen"
cd src/test/cpp/regression
export BUILDROOT=$(buildroot directory path)
make clean run IBUS=CACHED DBUS=CACHED  DEBUG_PLUGIN=STD SUPERVISOR=yes CSR=yes DEBUG_PLUGIN=no  COMPRESSED=no LRSC=yes AMO=yes REDO=0 DHRYSTONE=no LINUX_SOC=yes EMULATOR=../../../main/c/emulator/build/emulator.bin VMLINUX=$BUILDROOT/output/images/vmlinux DTB=$BUILDROOT/board/spinal/vexriscv_sim/rv32.dtb RAMDISK=$BUILDROOT/output/images/rootfs.cpio WITH_USER_IO=yes TRACE=no FLOW_INFO=no
```
I modified` VMLINUX=$BUILDROOT/output/images/Image` by `VMLINUX=$BUILDROOT/output/images/vmlinux`

It doesn't work, I have the following message:
```
./obj_dir/VVexRiscv
BOOT
*** VexRiscv BIOS ***
*** Supervisor ***
FAIL linux at PC=80000800 REF PC=800007fc REF I=6f time=3506
make: *** [makefile:264 : run] Erreur 255
```
Very nice video, very impressive!
How did you implement the VGA interface? did you need to develop a specific peripheral in the FPGA for PmOD VGA module interfacing?

Dolu1990, 2021-11-18:
I have to say, i haven't use that buildroot flow to generate images sinces ages (same for vexriscv.demo.LinuxGen), so , maybe try first with : 
https://github.com/SpinalHDL/VexRiscvRegressionData/tree/539398c1481203a51115b5f1228ea961f0ac9bd3/sim/linux/rv32ima
https://github.com/SpinalHDL/VexRiscvRegressionData/tree/539398c1481203a51115b5f1228ea961f0ac9bd3/sim/linux/emulator

which are precompiled images that i still use for automated regressions which test them on randomized VexRiscv configs. See Test_linux on : 
https://github.com/SpinalHDL/VexRiscv/runs/4209092497?check_suite_focus=true

> VMLINUX=$BUILDROOT/output/images/vmlinux

This isn't well named, i think you realy need the $BUILDROOT/output/images/Image, as that's what is used for regressions

> sbt "runMain vexriscv.demo.LinuxGen"

If it still doesn't work with the precompiled images, let me know, it may be a few missing optionnal feature to turn on.

> How did you implement the VGA interface?

I used : 
 https://www.digikey.ch/product-detail/de/digilent,-inc./410-345/1286-1171-ND/7560228?utm_adgroup=General&utm_source=google&utm_medium=cpc&utm_campaign=Smart%20Shopping_Product_Zombie%20SKU&utm_term=&productid=7560228&gclid=Cj0KCQiAkNiMBhCxARIsAIDDKNWYTn0VnybR27AkZzaN7-R1vSyK15l6cFUgmdZIDV9iHitMkZTupLgaAlJ5EALw_wcB
 
> did you need to develop a specific peripheral in the FPGA for PmOD VGA module interfacing?

Yes, but it isn't complicated, there is also a HDMI output for the ulx3s :)

sjthales, 2021-11-19:
It works. Thank you.