SpinalHDL/VexRiscv #223: Debug failed

Saqlinahamad, 2021-12-03:
Hi, 
I am using lattice propel tool and debugging RISCV Core with Lattice Board LFCPNX-100 in lattice propel SDK. 
While debugging I am getting following error for default program 
For the same board, I debugged default program just 10 days back,

Open On-Chip Debugger 0.10.0+dev-snapshot (2021-10-14-10:56) Licensed under GNU GPL v2 For bug reports, read http://openocd.org/doc/doxygen/bugs.html 0 5 FTUSB-0 Info : only one transport option; autoselect 'jtag' none separate Warn : Transport "jtag" was already selected adapter speed: 4000 kHz E:/Saqlain/propel_builder/2.1_Beta/rvc/src/cpu0.yaml Warn : Interface already configured, ignoring adapter speed: 4000 kHz adapter_nsrst_delay: 260 jtag_ntrst_delay: 250 Info : set servers polling period to 50ms QCoreApplication::applicationDirPath: Please instantiate the QApplication object first Device List: LFCPNX-100:0x010F4043
Info : clock speed 4000 kHz Info : TAP fpga_spinal.bridge does not have IDCODE Info : TAP auto0.tap does not have IDCODE Info : TAP auto1.tap does not have IDCODE Info : TAP auto2.tap does not have IDCODE Info : TAP auto3.tap does not have IDCODE Info : TAP auto4.tap does not have IDCODE Info : TAP auto5.tap does not have IDCODE Info : TAP auto6.tap does not have IDCODE Info : TAP auto7.tap does not have IDCODE Info : TAP auto8.tap does not have IDCODE Info : TAP auto9.tap does not have IDCODE Info : TAP auto10.tap does not have IDCODE Info : TAP auto11.tap does not have IDCODE Info : TAP auto12.tap does not have IDCODE Info : TAP auto13.tap does not have IDCODE Info : TAP auto14.tap does not have IDCODE Info : TAP auto15.tap does not have IDCODE Info : TAP auto16.tap does not have IDCODE Info : TAP auto17.tap does not have IDCODE Info : TAP auto18.tap does not have IDCODE Info : TAP auto19.tap does not have IDCODE

Warn : Unexpected idcode after end of chain: 21 0x00000000 Warn : Unexpected idcode after end of chain: 53 0x00000000 Warn : Unexpected idcode after end of chain: 85 0x00000000 Warn : Unexpected idcode after end of chain: 117 0x00000000 Warn : Unexpected idcode after end of chain: 149 0x00000000 Warn : Unexpected idcode after end of chain: 181 0x00000000 Warn : Unexpected idcode after end of chain: 213 0x00000000 Warn : Unexpected idcode after end of chain: 245 0x00000000 Warn : Unexpected idcode after end of chain: 277 0x00000000 Warn : Unexpected idcode after end of chain: 309 0x00000000 Warn : Unexpected idcode after end of chain: 341 0x00000000 Warn : Unexpected idcode after end of chain: 373 0x00000000 Warn : Unexpected idcode after end of chain: 405 0x00000000 Warn : Unexpected idcode after end of chain: 437 0x00000000 Warn : Unexpected idcode after end of chain: 469 0x00000000 Warn : Unexpected idcode after end of chain: 501 0x00000000 Warn : Unexpected idcode after end of chain: 533 0x00000000 Warn : Unexpected idcode after end of chain: 565 0x00000000 Warn : Unexpected idcode after end of chain: 597 0x00000000 Warn : Unexpected idcode after end of chain: 629 0x00000000 Error: double-check your JTAG setup (interface, speed, ...) Error: Trying to use configured scan chain anyway... Error: fpga_spinal.bridge: IR capture error; saw 0x00 not 0x01 Warn : Bypassing JTAG setup events due to errors Info : Listening on port 3333 for gdb connections Started by GNU MCU Eclipse Info : Listening on port 6666 for tcl connections Info : Listening on port 4444 for telnet connections Info : accepting 'gdb' connection on tcp/3333 Info : Halt timed out, wake up GDB. Error: timed out while waiting for target halted Info : TAP fpga_spinal.bridge does not have IDCODE Info : TAP auto0.tap does not have IDCODE Info : TAP auto1.tap does not have IDCODE Info : TAP auto2.tap does not have IDCODE Info : TAP auto3.tap does not have IDCODE Info : TAP auto4.tap does not have IDCODE Info : TAP auto5.tap does not have IDCODE Info : TAP auto6.tap does not have IDCODE Info : TAP auto7.tap does not have IDCODE Info : TAP auto8.tap does not have IDCODE Info : TAP auto9.tap does not have IDCODE Info : TAP auto10.tap does not have IDCODE Info : TAP auto11.tap does not have IDCODE Info : TAP auto12.tap does not have IDCODE Info : TAP auto13.tap does not have IDCODE Info : TAP auto14.tap does not have IDCODE Info : TAP auto15.tap does not have IDCODE Info : TAP auto16.tap does not have IDCODE Info : TAP auto17.tap does not have IDCODE Info : TAP auto18.tap does not have IDCODE Info : TAP auto19.tap does not have IDCODE Info : TAP auto0.tap does not have IDCODE

Warn : Unexpected idcode after end of chain: 22 0x00000000 Warn : Unexpected idcode after end of chain: 54 0x00000000 Warn : Unexpected idcode after end of chain: 86 0x00000000 Warn : Unexpected idcode after end of chain: 118 0x00000000 Warn : Unexpected idcode after end of chain: 150 0x00000000 Warn : Unexpected idcode after end of chain: 182 0x00000000 Warn : Unexpected idcode after end of chain: 214 0x00000000 Warn : Unexpected idcode after end of chain: 246 0x00000000 Warn : Unexpected idcode after end of chain: 278 0x00000000 Warn : Unexpected idcode after end of chain: 310 0x00000000 Warn : Unexpected idcode after end of chain: 342 0x00000000 Warn : Unexpected idcode after end of chain: 374 0x00000000 Warn : Unexpected idcode after end of chain: 406 0x00000000 Warn : Unexpected idcode after end of chain: 438 0x00000000 Warn : Unexpected idcode after end of chain: 470 0x00000000 Warn : Unexpected idcode after end of chain: 502 0x00000000 Warn : Unexpected idcode after end of chain: 534 0x00000000 Warn : Unexpected idcode after end of chain: 566 0x00000000 Warn : Unexpected idcode after end of chain: 598 0x00000000 Warn : Unexpected idcode after end of chain: 630 0x00000000 Warn : Unexpected idcode after end of chain: 662 0x00000000 Error: double-check your JTAG setup (interface, speed, ...) Error: Trying to use configured scan chain anyway... Error: fpga_spinal.bridge: IR capture error; saw 0x00 not 0x01 Warn : Bypassing JTAG setup events due to errors Error: timed out while waiting for target halted TARGET: fpga_spinal.cpu0 - Not halted Info : Halt timed out, wake up GDB. Error: timed out while waiting for target halted Assertion failed: target->state == TARGET_HALTED, file src/target/vexriscv.c, line 1059

fatalfeel, 2021-12-04:
http://fatalfeel.blogspot.com/2013/12/risc-v-on-arty-a7-35t.html?m=1
Check external jtag part make sure u are using dpinalhdl's openocd

fatalfeel, 2021-12-04:
The idcode number already in vecriscv. its in vexriscv import pack

Saqlinahamad, 2021-12-04:
Will you please help me in making sure about external JTAG part number

As I am new to this things 
I am just debugging RISCV Core in lattice propel SDK tool with provided board and lattice cable

Saqlinahamad, 2021-12-04:
[image]
[image]

Saqlinahamad, 2021-12-04:
I think it's GDB OpenOCD Debugging

fatalfeel, 2021-12-04:
openocd will auto get idcode~~~where u get openocd?

Saqlinahamad, 2021-12-04:
In Propel tool, 
We have option to debug through OpenOCD or GDB.

fatalfeel, 2021-12-04:
I mean the openocd git address

Saqlinahamad, 2021-12-04:
No, I haven't cloned this repo

fatalfeel, 2021-12-04:
before my exp. i use official openocd git://git.code.sf.net/p/openocd/code is failed and using SpinalHDL/openocd_riscv is ok!

Saqlinahamad, 2021-12-04:
The tool inbuilt is containing OpenOCD 
I am Debugging the FPGA board with GDB Openocd debugger 
As I debugged just 15-20 days back, 
Now 
I don't know why I can't able to debug,

fatalfeel, 2021-12-04:
try again https://github.com/SpinalHDL/openocd_riscv

Saqlinahamad, 2021-12-04:
Does it requires a build in linux OS 
or 
Cloning a directory? If Cloning a directory
In OpenOCD Setup, 
I require executable file OpenOCD.exe

fatalfeel, 2021-12-04:
in ubuntu~i think u have a lot road to go~~~but good luck first

Saqlinahamad, 2021-12-04:
No, I am working in Windows

Saqlinahamad, 2021-12-04:
C:/lscc/propel/2.1/SDK/Openocd/bin/openocd.exe 

If I cloned the Vexriscv Openocd 
What should be my Openocd application path?

Saqlinahamad, 2021-12-04:
[image]

Saqlinahamad, 2021-12-04:
[image]

fatalfeel, 2021-12-04:
u need upload to mediafire or google driver or we can not see it ~

Saqlinahamad, 2021-12-04:
https://photos.app.goo.gl/VjTnDC71higH2DUTA

Saqlinahamad, 2021-12-04:
Had shared the executable directory structure and path with the debug configuration I have currently

fatalfeel, 2021-12-04:
my work system is on ubuntu PC not simulator linux , so can not help you in windows.
in ubuntu, we use terminal window to run openocd, recommend u change to ubuntu.
the gcc we also build in ubuntu, not use any sdk tools

Saqlinahamad, 2021-12-05:
Yeah, 
I am working in ofc, and it will take some time to shift to Ubuntu, but I don't have much knowledge on Ubuntu also 
I am not able to locate this file (vexriscv.c) in my PC Directory structure 
What else can I be able to do in Windows platform only?

Saqlinahamad, 2021-12-05:
Is it a JTAG interface target issue or 
Board Issue ?

Dolu1990, 2021-12-05:
hmmm to me, it seems like some electrical issues. would that be possible ?
Do you have a logic analyser ? can you probe the jtag pins ?

Saqlinahamad, 2021-12-06:
Yeah, 
Had checked the JTAG pins with probes in oscilloscope
They are working fine.

Dolu1990, 2021-12-07:
Could you share a capture ? from the moment openocd boot ?

Saqlinahamad, 2021-12-07:
Hi @Dolu1990 , had captured the Debug of my SW Build over a FPGA board
Please reply me 
As soon as possible

tomverbeure, 2021-12-07:
@Saqlinahamad 
You need to be more specific about what you're trying to do and what you're trying to archieve.

It seems like you're using a version of OpenOCD that comes standard with the Lattice Propel Tool. The default version of OpenOCD does not support the VexRiscv. As @fatalfeel already wrote multiple times: you have to use the VexRiscv openocd version that you can find at https://github.com/spinalhdl/openocd_riscv.

In the picture that you're attaching, I see "RISCV_AHBLite_RPC_DRAM_Controller_SV...". I see a press release that shows that [Lattice Propel has its own RISC-V CPU](https://www.yahoo.com/now/lattice-propel-design-environment-supports-200000942.html). This makes me question: what design are you actually using? Are you using the Propel RISC-V CPU?

If so, you're asking your question in the wrong forum. This project is about the VexRiscv, a RISC-V CPU that has nothing to do with the Lattice RISC-V.

fatalfeel, 2021-12-08:
if you insist work on windows 
try download cygwin64 
http://www.cygwin.com
enable devel option for development gcc

build https://github.com/spinalhdl/openocd_riscv at windows
But I am not sure work or not~

Saqlinahamad, 2021-12-08:
Okay
But, I need to produce debug results with Lattice RISC-V only. 
Thanks

closed 2021-12-08 by Saqlinahamad