SpinalHDL/VexRiscv #224: Failed when apply nativeJtag for Altera DE0-Nano-SoC board

vietthanh85, 2021-12-10:
I have followed to https://github.com/SpinalHDL/VexRiscv/tree/master/doc/nativeJtag, but I need to modify for my Altera DE0-Nano-SoC board (Cyclone® V SE 5CSEMA4U23C6N device) and Altera Virtual JTAG. Unfortunately, it was not successful yet. The following is my modifications.

**Original (Xilinx Bscane2)**
```
[254]	val jtagCtrl = JtagTapInstructionCtrl()
[255]	val tap = jtagCtrl.fromXilinxBscane2(userId = 2)
[256]	jtagCtrl <> plugin.io.bus.fromJtagInstructionCtrl(ClockDomain(tap.TCK))
```
**Modification for Altera Virtual JTAG**
```
val tap = new VJTAG()
val jtagCtrl = tap.toJtagTapInstructionCtrl()
jtagCtrl <> plugin.io.bus.fromJtagInstructionCtrl(ClockDomain(tap.tck))
```
**Original: Add import for JtagTapInstructionCtrl**
```
import spinal.lib.com.jtag.JtagTapInstructionCtrl
```
**Modification: Add import for JtagTapInstructionCtrl and VJTAG**
```
import spinal.lib.com.jtag.JtagTapInstructionCtrl
import spinal.lib.blackbox.altera.VJTAG
```
Then, it generated verilog and hex files successfully.
I used them to make Quartus project.
I also added VJTAG module to Quartus project.
(Currently, I reused `vJTAG.qip`, `vJTAG.v` from this example: http://idle-logic.com/2012/04/15/talking-to-the-de0-nano-using-the-virtual-jtag-interface/)
For `toplevel.v`, I also removed the following lines
```
[43]	reg  tesic_tck,tesic_tms,tesic_tdi;
[44]	wire tesic_tdo;
[45]	reg  soc_tck,soc_tms,soc_tdi;
[46]	wire soc_tdo;
[47]
[48]	always @(*) begin
[49]		{soc_tck,  soc_tms,  soc_tdi  } = {tck,tms,tdi};
[50]		tdo = soc_tdo;
[51]	end
…
[56]	.io_jtag_tck(soc_tck),
[57]	.io_jtag_tdi(soc_tdi),
[58]	.io_jtag_tdo(soc_tdo),
[59]	.io_jtag_tms(soc_tms),
```

Then, the bitstream was loaded into the board successfully and Murax worked normally (LED blinked).

For debugging by OpenOCD, I modified as follows:
Create new file `altera-5csema4.cfg` in folder `openocd_riscv/tcl/cpld/` with following content (I have checked the corresponding device handbook for Cyclone V and found that the `expected-id` should be `0x02d010dd`):
```
# Altera Cyclone V 5CSEMA4
# see Cyclone V Device Handbook
# https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/hb/cyclone-v/cv_5v2.pdf
# Table 9-1: IDCODE Information for Cyclone V Devices
# Version     Part Number             Manuf. ID        LSB
# 0000        0010 1101 0000 0001     000 0110 1110    1
jtag newtap 5csema4 tap -expected-id 0x02d010dd -irlen 10
```
Then, I modified `usb_connect.cfg` to work with Altera USB Blaster instead of "Digilent USB Device":
```
set _CHIPNAME 5csema4

# adapter driver ftdi
# ftdi_device_desc "Digilent USB Device"
# ftdi_vid_pid 0x0403 0x6010
# ftdi_channel 0
# ftdi_layout_init 0x00e8 0x60eb
# ftdi_tdo_sample_edge falling
adapter driver usb_blaster
usb_blaster vid_pid 0x09fb 0x6010 0x09fb 0x6810
usb_blaster lowlevel_driver ublast2
usb_blaster firmware /home/thanh/intelFPGA_lite/21.1/quartus/linux64/blaster_6810.hex

reset_config none
adapter speed 5000

# source [find cpld/xilinx-xc7.cfg]
# source [find cpld/jtagspi.cfg]
source [find tcl/cpld/altera-5csema4.cfg]
source [find tcl/cpld/jtagspi.cfg]

# set TAP_NAME xc7.tap
set TAP_NAME 5csema4.tap
```
I did not modify `soc_init.cfg`.
For other steps, I followed to the reference.

Here is the result when I tried to debug with OpenOCD:
```shell
$ cd openocd_riscv
$ openocd -c "set CPU0_YAML ../VexRiscv/cpu0.yaml" \
-f tcl/interface/usb_connect.cfg \
-f tcl/interface/soc_init.cfg
Open On-Chip Debugger 0.11.0+dev-02572-ga985acb34-dirty (2021-12-04-15:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
../VexRiscv/cpu0.yaml
Info : only one transport option; autoselect 'jtag'
5csema4.tap
Info : set servers polling period to 50ms
Info : Altera USB-Blaster II found (Firm. rev. = 1.39)
Info : This adapter doesn't support configurable speed
Info : JTAG tap: 5csema4.tap tap/device found: 0x02d010dd (mfg: 0x06e (Altera), part: 0x2d01, ver: 0x0)
Info : JTAG tap: auto0.tap tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x4)
Warn : AUTO auto0.tap - use "jtag newtap auto0 tap -irlen 4 -expected-id 0x4ba00477"
Info : starting gdb server for saxon.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
requesting target halt and executing a soft reset
 done
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : New GDB Connection: 1, Target saxon.cpu0, state: halted
Info : JTAG tap: 5csema4.tap tap/device found: 0x02d010dd (mfg: 0x06e (Altera), part: 0x2d01, ver: 0x0)
Info : JTAG tap: auto0.tap tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x4)
Error: timed out while waiting for target halted
TARGET: saxon.cpu0 - Not halted
```
GDB client terminal:
```
$ riscv64-unknown-elf-gdb  /home/thanh/SpinalHDL_VexRiscv/VexRiscvSocSoftware/projects/murax/demo/build/demo.elf 
GNU gdb (SiFive GDB 9.1.0-2020.08.2) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/sifive/freedom-tools/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/thanh/SpinalHDL_VexRiscv/VexRiscvSocSoftware/VexRiscvSocSoftware/projects/murax/demo/build/demo.elf...
(gdb) target extended-remote localhost:3333
Remote debugging using localhost:3333
0x00000000 in ?? ()
(gdb) load
warning: while parsing target memory map (at line 2): Required element <memory> is missing
Loading section .vector, size 0x130 lma 0x80000000
Loading section .memory, size 0x358 lma 0x80000130
Start address 0x80000000, load size 1160
Transfer rate: 377 KB/sec, 580 bytes/write.
(gdb) monitor reset halt
JTAG tap: 5csema4.tap tap/device found: 0x02d010dd (mfg: 0x06e (Altera), part: 0x2d01, ver: 0x0)
JTAG tap: auto0.tap tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x4)
timed out while waiting for target halted
TARGET: saxon.cpu0 - Not halted
```
The problem is that: CPU was not halted after I sent `monitor reset halt` from gdb client.
So, I cannot debug anything the application :(
I guess that the OpenOCD configurations to connect to USB Blaster and the board might be not correct, but I am not sure how to configure it correctly.
Do you have any idea about what the wrong steps are?
Your help would be very appreciated!
Many thanks!

Dolu1990, 2021-12-14:
Hi,
I would say, to debug it you would need to check what is happning on the VJTAG using signaltap (or another logic analyser), and see if it does something when you start openocd.

That way we can be sure which way to look at, as there is quite many things involved to get things to work ^^

vietthanh85, 2021-12-14:
> Hi, I would say, to debug it you would need to check what is happning on the VJTAG using signaltap (or another logic analyser), and see if it does something when you start openocd.
> 
> That way we can be sure which way to look at, as there is quite many things involved to get things to work ^^

@Dolu1990 Thank you. I've got your point.

tomverbeure, 2021-12-14:
> I guess that the OpenOCD configurations to connect to USB Blaster and the board might be not correct, but I am not sure how to configure it correctly. Do you have any idea about what the wrong steps are? Your help would be very appreciated! 

OpenOCD (including the VexRiscv version of OpenOCD) doesn't have a generic way to deal with Intel's virtual JTAG. To my knowledge, there is only one CPU that supports virtual JTAG in OpenOCD and that's OpenRISC. (See [here](https://github.com/openocd-org/openocd/blob/master/src/target/openrisc/or1k_tap_vjtag.c).)

Support of virtual JTAG for the VexRiscv requires a similar amount of work in OpenOCD. Personally, I simply [use general purpose GPIOs and the SpinalHDL JTAG TAP](https://tomverbeure.github.io/2021/07/18/VexRiscv-OpenOCD-and-Traps.html) to connect a debugger to a VexRiscv CPU.

Tom

Dolu1990, 2021-12-15:
If i remember well, somebody got it to work with the vjtag from altera, but i realy don't remember anything else XD

vietthanh85, 2021-12-15:
> OpenOCD (including the VexRiscv version of OpenOCD) doesn't have a generic way to deal with Intel's virtual JTAG. To my knowledge, there is only one CPU that supports virtual JTAG in OpenOCD and [...]

Thank @tomverbeure . By the way, I've just found this blog https://koyamanx.github.io/ck-dev/blog/my_riscv_debug_feature_part1. He tried to debug RISC-V cores on Altera boards by OpenOCD using VJTAG.
He also ported it from OpenRISC (or1k_tap_vjtag.c). Unfortunately, only Japanese is available. I am following to this blog post through a lot of things need to be modified and it's still far from the success. Hope this helps in case of you have a similar work to do.

tomverbeure, 2021-12-15:
Nice find!
The number of changes that are specific to virtual JTAG are very reasonable: https://github.com/koyamanX/riscv-openocd/commit/ad03470a2b4d3f0fadd9e1944445ac9193487d37. 

Tom

LYWalker, 2022-01-31:
@vietthanh85 Any luck getting this to work in the end? Trying to have a go myself

vietthanh85, 2022-02-01:
> @vietthanh85 Any luck getting this to work in the end? Trying to have a go myself
@LYWalker not yet.
https://koyamanx.github.io/ck-dev/blog/my_riscv_debug_feature_part1
https://github.com/koyamanX/riscv-debug
This is not a complete guideline. A lot of TODOs to get it working.
I am thinking to use another solution from LiteX:
https://github.com/jevinskie/litex/blob/jev/main/litex/soc/cores/jtag.py

LYWalker, 2022-11-04:
I can confirm that I have VJTAG working, I'll create a pull request soon. It has edits in the openocd port as well, as you can imagine.

Dolu1990, 2022-11-06:
@LYWalker Nice thanks :D

LYWalker, 2022-11-07:
@Dolu1990 
See these pull requests:
https://github.com/SpinalHDL/SpinalHDL/pull/950
https://github.com/SpinalHDL/openocd_riscv/pull/28
#276 

Let me know if you have any suggestions. I've tested it successfully on the DE0 board.